carlonluca/pot #12: Garbage on video changed

desert, 2015-02-23:
When using latest 4.3.0 version and Qt 5.4 (from the tar in your blog) i get garbage on screen when changing the video it looks as if there is a buffer which is not zeroed so we get noise.

     import QtQuick 2.3
     import QtQuick.Controls 1.2
     import QtQuick.Window 2.2
     import QtMultimedia 5.0

     ApplicationWindow {
     id: app
     title: qsTr("Hello World")
     width: 640
     height: 480
     visible: true
     property var videos: ["/home/pi/test.mp4", "/home/pi/test1.mp4", "/home/pi/test2.mp4", "/home/pi/test3.mp4", "/home/pi/este-no-existe.mp4"]
     property int index:0
     Video{
        id: vid
        width: 400
        height: 400

        anchors.fill: parent
        source: videos[index]
        autoPlay: true
        onStopped: {
            var i = app.index
            i = (i+1)%videos.length
            index=i
        }
       }
      }

carlonluca, 2015-02-23:
Yes, try the no_tearing branch.

carlonluca, 2015-03-03:
Did you have a look at that branch. Does it fix?

desert, 2015-03-03:
I have not, i do have and environment for cross compiling Qt5 and i cross compile apps on a daily basis, it is just that i cannot cross compile your project. I have interest on this, so if you would write a post or meetup with me on IRC, i can fix my cross compiling and i can help fixing stuff.

carlonluca, 2015-03-03:
I posted a few guides on the blog. This is the latest: http://thebugfreeblog.blogspot.it/2014/12/updates-on-hardware-accelerated-qt.html.

carlonluca, 2015-03-28:
Can someone confirm that no_tearing branch no more shows artifacts when video starts?

desert, 2015-03-29:
I have used several videos with the binaries provided by @Nightmare204 but all of them have garbage at the beginning, either gray screens or artifacts. He says we need to fix the videos with a black frame but all of the videos display fine in all other players, I still think this is a bug.

Maybe the frames in the lib need to be internally set to pure black at the very beginning in order to avoid artifacts and avoid "fixing" videos with additional black frames.I have used several videos with the binaries provided by @Nightmare204 but all of them have garbage at the beginning, either gray screens or artifacts. He says we need to fix the videos with a black frame but all of the videos display fine in all other players, I still think this is a bug.

Maybe the frames in the lib need to be internally set to pure black at the very beginning in order to avoid artifacts and avoid "fixing" videos with additional black frames.

carlonluca, 2015-03-29:
I uploaded a fix for this. Check the commit logs and let me know.

desert, 2015-03-30:
I used the no_tearing branch, compiled my own Qt 5.4.1 and I still see a gray screen and artifacts. Commit: 59523ef4b59f89590b126701f4fd1b2e98d591ba

desert, 2015-03-30:
I can also see this gray box when not playing, the trace is (this is my own compiled version)

```
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
15:12:51.676 VERBOSE: Instantiating player service...
15:12:51.678 VERBOSE: Instantiating QMediaService...
15:12:51.682 DEBUG: Entering: OpenMAXILPlayerControl::OpenMAXILPlayerControl(QObject*).
15:12:51.683 VERBOSE: Metadata update requested.
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
15:12:51.688 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMediaPlayer(QMediaPlayer*).
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
15:12:51.691 DEBUG: Entering: virtual QMediaPlayer::State OpenMAXILPlayerControl::state() const.
15:12:51.692 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
15:12:51.693 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
15:12:51.707 DEBUG: Entering: virtual void OpenMAXILVideoRendererControl::setSurface(QAbstractVideoSurface*).
15:12:51.758 DEBUG: Entering: virtual void OpenMAXILPlayerControl::stop().
15:12:51.760 VERBOSE: Stop
15:12:51.761 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMedia(const QMediaContent&, QIODevice*).
15:12:51.763 DEBUG: Media: /home/pi/test.mp4.
15:12:51.764 DEBUG: setMedia thread is: 0x868a88.
15:12:51.764 VERBOSE: Deferring setMedia()...
15:12:51.807 DEBUG: Renderer thread is: 0x928230.
15:12:51.808 VERBOSE: Processing post setMedia()...
15:12:51.809 DEBUG: Entering: void OpenMAXILPlayerControl::setMediaInt(const QMediaContent&).
15:12:51.810 VERBOSE: Stop
15:12:51.811 VERBOSE: Entering: bool OMX_MediaProcessor::setFilenameInt(QString, OMX_TextureData*&).
15:12:51.812 VERBOSE: Opening file:///home/pi/test.mp4...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/test.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41
    creation_time   : 2014-03-11 18:56:36
  Duration: 00:03:59.68, start: 0.000000, bitrate: 5584 kb/s
    Stream #0:0(eng): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv), 1280x720 [SAR 1:1 DAR 16:9], 5385 kb/s, 29.97 fps, 29.97 tbr, 29970 tbn, 59.94 tbc (default)
    Metadata:
      creation_time   : 2014-03-11 18:56:36
      handler_name    : Mainconcept MP4 Video Media Handler
      encoder         : AVC Coding
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 189 kb/s (default)
    Metadata:
      creation_time   : 2014-03-11 18:56:36
      handler_name    : Mainconcept MP4 Sound Media Handler
15:12:52.030 VERBOSE: Copy metatada...
15:12:52.031 VERBOSE: MetaData
15:12:52.032 VERBOSE: Key: major_brand, Value: mp42.
15:12:52.033 VERBOSE: Key: minor_version, Value: 0.
15:12:52.033 VERBOSE: Key: compatible_brands, Value: mp42mp41.
15:12:52.034 VERBOSE: Key: creation_time, Value: 2014-03-11 18:56:36.
15:12:52.036 VERBOSE: Initializing OMX clock...
15:12:52.045 VERBOSE: Opening video using OMX...
15:12:52.047 VERBOSE: Closing video...
15:12:52.048 VERBOSE: Deinit of 0x9d5f90
15:12:52.049 VERBOSE: Deinit of 0x9d5c20
15:12:52.050 VERBOSE: m_omx_render is 0x9d5dd8 freeing 0
15:12:52.050 VERBOSE: Deinit of 0x9d5dd8
15:12:52.051 VERBOSE: Generating texture of size (1280, 720).
15:12:52.052 INFORMATION: Instantiating texture data...
15:12:52.071 INFORMATION: Creating EGLImageKHR...
15:12:52.072 VERBOSE: EGL image 13 created...
15:12:52.073 VERBOSE: Creating OMX_TextureData...
15:12:52.074 INFORMATION: Instantiating texture data...
[... 41 lines cut ...]
15:12:54.340 VERBOSE: Component renderer: 10356c0.
15:12:54.388 WARNING: One frame couldn't be shown.
15:12:54.395 WARNING: One frame couldn't be shown.
15:12:54.421 WARNING: One frame couldn't be shown.
15:12:54.429 WARNING: One frame couldn't be shown.
15:12:54.455 WARNING: One frame couldn't be shown.
15:12:54.463 WARNING: One frame couldn't be shown.
15:12:54.481 WARNING: One frame couldn't be shown.
15:12:54.506 WARNING: One frame couldn't be shown.
15:12:54.515 WARNING: One frame couldn't be shown.
15:12:54.540 WARNING: One frame couldn't be shown.
15:12:54.548 WARNING: One frame couldn't be shown.
15:12:54.574 WARNING: One frame couldn't be shown.
15:12:54.581 WARNING: One frame couldn't be shown.
15:12:54.609 WARNING: One frame couldn't be shown.
15:12:54.615 WARNING: One frame couldn't be shown.
15:12:54.640 WARNING: One frame couldn't be shown.
15:12:54.666 WARNING: One frame couldn't be shown.
15:12:54.674 WARNING: One frame couldn't be shown.
15:12:54.699 WARNING: One frame couldn't be shown.
15:12:54.708 WARNING: One frame couldn't be shown.
15:12:54.733 WARNING: One frame couldn't be shown.
15:12:54.750 WARNING: One frame couldn't be shown.
15:12:54.758 WARNING: One frame couldn't be shown.
```

carlonluca, 2015-03-30:
That commit is old. Check the logs. Use the latest in no_tearing. Master is merged anyway, so you can also use latest master.

desert, 2015-03-31:
With master Im getting linker errors:
```
/opt/rpi/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../lib/gcc/arm-linux-gnueabihf/4.7.2/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lssl
/opt/rpi/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../lib/gcc/arm-linux-gnueabihf/4.7.2/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lcrypto
collect2: error: ld returned 1 exit status
make: *** [libPiOmxTextures.so.4.4.0] Error 1
```

What *-dev libs do you need into the sysroot?

desert, 2015-03-31:
Well I just commented out those linked libs and it runs.

I see no artifacts, but I still see a gray flash on every video and a neverending console output  like this which takes a lot CPU usage:

```
17:22:43.466 DEBUG: CDVDPlayerAudio::Decode dts:13482667 pts:13482667 size:480
17:22:43.468 DEBUG: CDVDPlayerAudio::Decode dts:13504000 pts:13504000 size:494
17:22:43.470 DEBUG: CDVDPlayerAudio::Decode dts:13525333 pts:13525333 size:503
17:22:43.471 DEBUG: CDVDPlayerAudio::Decode dts:13546667 pts:13546667 size:498
17:22:43.473 DEBUG: CDVDPlayerAudio::Decode dts:13568000 pts:13568000 size:503
17:22:43.475 DEBUG: CDVDPlayerVideo::Decode dts:12746079 pts:12746079 cur:12746079, size:1872
17:22:43.477 DEBUG: CDVDPlayerAudio::Decode dts:13589333 pts:13589333 size:504
17:22:43.478 DEBUG: Setting texture 1.
17:22:43.480 DEBUG: CDVDPlayerAudio::Decode dts:13610667 pts:13610667 size:515
17:22:43.481 DEBUG: CDVDPlayerAudio::Decode dts:13632000 pts:13632000 size:519
17:22:43.507 DEBUG: CDVDPlayerVideo::Decode dts:12779446 pts:12779446 cur:12779446, size:136
17:22:43.511 DEBUG: Setting texture 2.
17:22:43.541 DEBUG: Setting texture 3.
17:22:43.571 DEBUG: Setting texture 4.
17:22:43.579 DEBUG: CDVDPlayerVideo::Decode dts:12812813 pts:12946280 cur:12946280, size:4055
17:22:43.581 DEBUG: CDVDPlayerVideo::Decode dts:12846180 pts:12846180 cur:12846180, size:164
17:22:43.583 DEBUG: CDVDPlayerVideo::Decode dts:12879546 pts:12879546 cur:12879546, size:139
17:22:43.596 DEBUG: CDVDPlayerVideo::Decode dts:12912913 pts:12912913 cur:12912913, size:163
17:22:43.603 DEBUG: Setting texture 1.
17:22:43.608 DEBUG: CDVDPlayerVideo::Decode dts:12946280 pts:13079746 cur:13079746, size:1803
17:22:43.632 DEBUG: Setting texture 2.
17:22:43.640 DEBUG: CDVDPlayerVideo::Decode dts:12979646 pts:12979646 cur:12979646, size:132
17:22:43.646 DEBUG: COMXAudio::ApplyVolume - Volume=1.00 (* 1.00 * 1.00)

17:22:43.647 DEBUG: CDVDPlayerAudio::Decode dts:13653333 pts:13653333 size:498
17:22:43.649 DEBUG: CDVDPlayerAudio::Decode dts:13674667 pts:13674667 size:505
17:22:43.650 DEBUG: CDVDPlayerAudio::Decode dts:13696000 pts:13696000 size:511
17:22:43.652 DEBUG: CDVDPlayerAudio::Decode dts:13717333 pts:13717333 size:493
17:22:43.653 DEBUG: CDVDPlayerAudio::Decode dts:13738667 pts:13738667 size:516
17:22:43.655 DEBUG: CDVDPlayerAudio::Decode dts:13760000 pts:13760000 size:508
17:22:43.657 DEBUG: CDVDPlayerAudio::Decode dts:13781333 pts:13781333 size:506
17:22:43.658 DEBUG: CDVDPlayerAudio::Decode dts:13802667 pts:13802667 size:503
17:22:43.663 DEBUG: Setting texture 2.
17:22:43.693 DEBUG: Setting texture 3.
17:22:43.723 DEBUG: Setting texture 4.
17:22:43.743 DEBUG: CDVDPlayerVideo::Decode dts:13013013 pts:13013013 cur:13013013, size:136
17:22:43.755 DEBUG: Setting texture 1.
17:22:43.775 DEBUG: CDVDPlayerVideo::Decode dts:13046380 pts:13046380 cur:13046380, size:156
17:22:43.785 DEBUG: Setting texture 2.
17:22:43.804 DEBUG: COMXAudio::ApplyVolume - Volume=1.00 (* 1.00 * 1.00)

17:22:43.805 DEBUG: CDVDPlayerAudio::Decode dts:13824000 pts:13824000 size:506
17:22:43.807 DEBUG: CDVDPlayerVideo::Decode dts:13079746 pts:13213213 cur:13213213, size:1039
17:22:43.809 DEBUG: CDVDPlayerAudio::Decode dts:13845333 pts:13845333 size:505
17:22:43.811 DEBUG: CDVDPlayerAudio::Decode dts:13866667 pts:13866667 size:506
17:22:43.812 DEBUG: CDVDPlayerAudio::Decode dts:13888000 pts:13888000 size:499
17:22:43.814 DEBUG: CDVDPlayerAudio::Decode dts:13909333 pts:13909333 size:491
17:22:43.815 DEBUG: Setting texture 3.
17:22:43.817 DEBUG: CDVDPlayerAudio::Decode dts:13930667 pts:13930667 size:494
17:22:43.818 DEBUG: CDVDPlayerAudio::Decode dts:13952000 pts:13952000 size:514
17:22:43.820 DEBUG: CDVDPlayerAudio::Decode dts:13973333 pts:13973333 size:516
17:22:43.845 DEBUG: Setting texture 4.
```

carlonluca, 2015-04-01:
You are missing libs. Logs can be disabled with a macro. Look at the Light logger header for more info.

Nightmare204, 2015-04-01:
@carlonluca I have tested the master branch as it has the latest commit and I can assure the video start improved a lot!!!
Not only I don't get garbage, no halos or artifacts, but also the time between videos is lower (like 1 sec)...
I still get some lost frames in a pair of [Videos](https://www.dropbox.com/home/Public/Videos); I would appreciate if you confirm it's a problem of the videos itself or if there's something that can be improved in PiOmxTextures.

@carlonluca I had just remember you disabled the position function in the No_Tearing branch and that seems to be unimplemented after the merge with the master branch... Do I open another Issue?

desert, 2015-04-01:
@Nightmare204 I just compiled master, I tried to test your videos but the link is broken.

carlonluca, 2015-04-01:
@Nightmare204 I can't get the videos too. Ah yes, the position implementation was hardlocking for some reason. Yes, open a quick issue. I'll have a closer look, thanks.

carlonluca, 2015-04-01:
@desert Can you provide the video that shows the gray flash?

Nightmare204, 2015-04-01:
Please try this [Link](https://www.dropbox.com/sh/bwprnaz4aq2jfhq/AACR7siMhquB_wbIFD0mzs5Za?dl=0) for the videos...

carlonluca, 2015-04-03:
I guess this bug is closed then. Unless @desert can provide a video that shows the issue. In case just reopen this issue.

closed 2015-04-03 by carlonluca