carlonluca/pot #14: Raspberry Pi 2 support?

desert, 2015-03-03:
Hi Luca, 

Is the 4.3.0 working on the Pi 2?

If not, and it's because of lack of hardware to test, i can provide one or two for you.

carlonluca, 2015-03-03:
Hello,
I'm sorry, as you say I don't have the hardware so I have no idea if and how it works. I'm sorry but I have many projects at the moment, so I can't guarantee time to work on it.
Did you try to use it?

Nightmare204, 2015-03-03:
I have tested it in both Raspberry Pi B+ and Raspberry Pi 2, with the latest kernel (just did rpi-update yesterday on both)... and the POC works normal for hardware accelerated videos.
Now trying to make a playlist loop... that's another story being developed in "Memory leaks #13"

carlonluca, 2015-03-03:
So if it is ok for you desert I'd close this issue. Pi 2 seems to be "supported".

desert, 2015-03-03:
Yes, my  v2 will come later this week. so let's close it and if I find anything i will open another issue.

closed 2015-03-03 by carlonluca