carlonluca/pot #15: Simultaneus Videos

Nightmare204, 2015-03-28:
Master branch allows to play videos simultaneously by using multiple players, but still has the memory leak problem...
No_Tearing branch allows **just one** media player at a time.

Simultaneous videos are not always neccesary but  that would allow useful components like one video fading into another, video comparing, video chat, etc...

carlonluca, 2015-08-10:
Should be available in the new version 4.5.0.

Nightmare204, 2015-08-31:
I just forgot to mention here that I had tested version 4.5.0 with more than 2 videos at the same time without problem (4 actually).
Of course the amount of simultaneous videos is proportional to the memory split dedicated to GPU and also to the frame size.
No problem with different video frame size at all.

However some test showed me that pausing one video or stopping it, sometimes the other videos also received the signal. I'll reopen if I can replicate this behaviour again.

Nightmare204, 2016-01-12:
@carlonluca Is there some limit on simultaneous videos? Today I tested with 5 and just 4 of them reproduced. Is this related with the buffers available?

I tested overclocking my Pi2 and giving a lot of Memory to GPU and I can always only get 4 simultaneous videos.

Of course I know this isn't interesting for UHD or HD videos, but I'm using low quality videos 320x180

carlonluca, 2016-01-12:
Not that I know. Only limitation I know is that GPU memory must be sufficient. Each player allocates 4 buffers, so allocating 5 players results in 20 buffers and textures. Can't think of other limitations.

Nightmare204, 2016-01-12:
From the 5th player or more I only get:
```
V:PortSettingsChanged: 320x180@25.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
17:15:23.215 VERBOSE: Entering: bool COMXVideo::SetVideoEGL().
17:15:23.222 DEBUG: Trying to enable to output port...
17:15:23.222 VERBOSE: Creating buffers for 4 images.
17:15:23.223 INFORMATION: Buffer created 0x6d221b00
17:15:23.223 INFORMATION: Buffer created 0x6d241060
17:15:23.224 INFORMATION: Buffer created 0x6d2223a0
17:15:23.224 INFORMATION: Buffer created 0x6d222410
17:15:23.224 VERBOSE: Component renderer: 6d221c90.
```

Normally, if the video where played I would get the following logs, at the end of the play:
```
VERBOSE: EOS in FillBufferDone cb.
VERBOSE: Media status changing from MEDIA_STATUS_LOADED to MEDIA_STATUS_END_OF_MEDIA...
VERBOSE: Stopping OMX clock...
VERBOSE: State changing from STATE_PLAYING to STATE_STOPPED...
```

I'm giving 800Mbts to GPU, and the same issue happens if I split only 256Mbts... 
Checking with htop shows there isn't high CPU load...

And my final test was to use different applications, but If I start 3 players in one, only one more can be started in the other program, and so on... total players always is maximum 4...

Could it be that they are all sharing even the Buffers due to the 
```
    QCoreApplication::setAttribute(Qt::AA_ShareOpenGLContexts, true);
```
???

carlonluca, 2016-01-12:
I don't see anything clearly wrong from the logs. Can you provide full logs and a sample QML as simple as possible reproducing the bug?

Nightmare204, 2016-01-13:
The file attached as a simple project showing this bug, and 2 log files, but there aren't logs that evidence the behavior (only 4 players start) ... As you will see I just compiled POT again today.

[VideoTests.zip](https://github.com/carlonluca/pi/files/88182/VideoTests.zip)

I also upload my test video...
[180p_SONY.mp4.zip](https://github.com/carlonluca/pi/files/88190/180p_SONY.mp4.zip)

Nightmare204, 2016-01-13:
I tested the qml with qmlscene and in this case (sometimes) 5 players start... but never 6

davidpez1019, 2016-02-09:
Hi Luca, actually i'm working with @Nightmare204, We tested the new build, but I only can reproduce 6 videos simultaneously the 7 video fails when I try reproduce it, this happens when the program is running with qmlscene, running the program from Qt only reproduce 4 or eventually 5 videos at max. 

In other side when we try do streaming of a .3gp video it fails, we tested with .mp4 and .sdp videos, the streaming for this run fine.