carlonluca/pot #16: Support Audio via 3.5mm jack

Nightmare204, 2015-03-28:
Right now PiOmxTextures plays audio only through HDMI, even if you try to force it via 3.5mm ('headphone') jack configuring it with raspi-config.
It would be nice if the audio output could be selected or followed the configuration of the raspberry.

desert, 2015-03-29:
Yes, specially since there are some HMDI to VGA converters, and you loose audio.

carlonluca, 2015-04-01:
What about both?

Nightmare204, 2015-04-04:
If audio comes out via both (HDMI and Jack) at the same time I don't think that would be a problem.
For those cases where you lose the audio through HDMI because of a converter this could be a solution...

carlonluca, 2015-04-05:
#4591c30 should solve this issue. Reopen if it is not fixing this.

closed 2015-04-05 by carlonluca

Nightmare204, 2015-04-06:
It's working Perfectly... Great work @carlonluca