carlonluca/pot #18: Build

inadsan, 2015-04-06:
Hello,
With the previous version everything worked well for me.
I have a question about the building system.

I compiled everything and piomxtextures_app working properly.
From a QML and qmlscene me now does not work.
I have the built and in place piomxtextures_qt_driver.
Is the patch qtmultimedia needed?

With patch:
12:36:09.836 VERBOSE: Instantiating player service...
12:36:09.838 VERBOSE: Instantiating QMediaService...
12:36:09.843 DEBUG: Entering: OpenMAXILPlayerControl::OpenMAXILPlayerControl(QObject*).
12:36:09.845 VERBOSE: Metadata update requested.
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
12:36:09.852 DEBUG: Entering: virtual QMediaPlayer::State OpenMAXILPlayerControl::state() const.
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
12:36:09.856 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
12:36:09.876 DEBUG: Entering: virtual void OpenMAXILVideoRendererControl::setSurface(QAbstractVideoSurface*).

No patch:
12:30:40.441 VERBOSE: Instantiating player service...
12:30:40.443 VERBOSE: Instantiating QMediaService...
12:30:40.449 DEBUG: Entering: OpenMAXILPlayerControl::OpenMAXILPlayerControl(QObject*).
12:30:40.451 VERBOSE: Metadata update requested.
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
12:30:40.460 DEBUG: Entering: virtual QMediaPlayer::State OpenMAXILPlayerControl::state() const.
12:30:40.462 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
12:30:40.463 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
12:30:40.481 DEBUG: Entering: virtual void OpenMAXILVideoRendererControl::setSurface(QAbstractVideoSurface*).
12:30:40.483 DEBUG: Entering: virtual void OpenMAXILPlayerControl::stop().
12:30:40.485 VERBOSE: Stop
12:30:40.487 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMedia(const QMediaContent&, QIODevice*).
12:30:40.488 DEBUG: Media: /content/VIDEO.mp4.
12:30:40.490 DEBUG: setMedia thread is: 0x1a78050.
12:30:40.491 VERBOSE: Deferring setMedia()...

Any thoughts?
thank you very much

carlonluca, 2015-04-06:
Yes, patching qt multimedia is still needed.

carlonluca, 2015-04-06:
But it is not necessary to build inside qt sources and link to an external shared lib.

inadsan, 2015-04-06:
ok, is necessary but not the last part of building libopenmaxilmediaplayer.so
which is made from your project.
Have I understood correctly?
I'll keep looking to see what my problem.
Thank You

carlonluca, 2015-04-06:
Patch qt multimedia, build it and then build the rest of PiOmxTextures, including the driver.

closed 2015-04-06 by carlonluca

dimovnike, 2015-04-08:
Im trying to compile it too, but the patched qtmultimedia doesn't compile, it says about missing openmaxil.pro file. What am I doing wrong? Thank you.

inadsan, 2015-04-08:
removes the last part of the patch or do not use the define + = raspberry to build qtmultimedia
Now the driver is constructed by qt outside the project.

carlonluca, 2015-04-08:
Yes, #20 reports that. I'll fix this.