carlonluca/pot #20: build instructions not working (not possible to build)

dimovnike, 2015-04-08:
the build instructions on the blog are not working, i tried different order of operations but still no luck. Any chance something is missing there?

dimovnike, 2015-04-08:
in the end it built some how, but I don't know what exactly i did. Looks like you need to build qtmultimedia with gstreamer first, then build piomxtextures_qt_driver, make install it, and then patch the qtmuiltimedia, run qmake with -nocache, clean it and rebuild.

carlonluca, 2015-04-08:
Yes, the qtmultimedia patch tries to build the backend, but this is not the case anymore with the new build procedure. I'll fix that.

carlonluca, 2015-04-08:
Should be fixed now. Reopen if you still encounter issue while building.

closed 2015-04-08 by carlonluca