carlonluca/pot #21: muting a video from QML crashes the app

dimovnike, 2015-04-08:
the app exits when mute property is used from QML, the app exits with:

06:29:09.189 DEBUG: Entering: virtual QMediaPlayer::State OpenMAXILPlayerControl::state() const.
06:29:09.190 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
06:29:09.191 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
06:29:09.276 DEBUG: Entering: virtual void OpenMAXILVideoRendererControl::setSurface(QAbstractVideoSurface*).
06:29:12.309 DEBUG: Entering: virtual bool OpenMAXILPlayerControl::isMuted() const.

desert, 2015-04-08:
I tried to compile latest to test this and Im getting crashes regarding
```../piomxtextures_src/omxplayer_lib/SubtitleRenderer.cpp:428:33: error: ‘DISPMANX_STEREOSCOPIC_MONO’ was not declared in this scope```

What was changed?

carlonluca, 2015-04-08:
@desert That is not a crash. That is a compiler error related to outdated headers. Update your firmware.

desert, 2015-04-08:
Sure, bad word,  my bad.  But you just gave me the answer. I will update my firmware.

carlonluca, 2015-04-08:
@dimovnike I just tested the mute API and I didn't get any crash. I added a mute button in POCPlayer if you want to try that. Seemed to work for me.

carlonluca, 2015-04-08:
Btw, volume was unimplemented instead for some reason... I quickly added that as well.

desert, 2015-04-09:
I was using the POC player now named piomxtextures_app, somehow by pushing randome keys too fast it crashes. I say this because y do not see any exit button in main.qml my log is:
```
20:57:43.165 VERBOSE: Opening big_buck_bunny_1080p_h264.mov...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'big_buck_bunny_1080p_h264.mov':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: isommp42
    creation_time   : 2014-11-25 08:00:05
  Duration: 00:00:13.00, start: 0.000000, bitrate: 640 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], 444 kb/s, 25 fps, 25 tbr, 25 tbn, 50 tbc (default)
    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 191 kb/s (default)
    Metadata:
      creation_time   : 2014-11-25 08:00:05
      handler_name    : IsoMedia File Produced by Google, 5-11-2011
20:57:43.573 VERBOSE: Copy metatada...
20:57:43.574 VERBOSE: MetaData
20:57:43.575 VERBOSE: Key: major_brand, Value: mp42.
20:57:43.576 VERBOSE: Key: minor_version, Value: 0.
20:57:43.577 VERBOSE: Key: compatible_brands, Value: isommp42.
20:57:43.578 VERBOSE: Key: creation_time, Value: 2014-11-25 08:00:05.
20:57:43.579 VERBOSE: Initializing OMX clock...
20:57:43.587 VERBOSE: Opening video using OMX...
20:57:43.588 VERBOSE: Generating texture of size (1280, 720).
20:57:43.589 INFORMATION: Freeing KHR image...
20:57:43.590 INFORMATION: Freeing texture...
20:57:43.591 INFORMATION: Freeing KHR image...
20:57:43.592 INFORMATION: Freeing texture...
20:57:43.594 INFORMATION: Freeing KHR image...
20:57:43.595 INFORMATION: Freeing texture...
20:57:43.596 INFORMATION: Freeing KHR image...
20:57:43.597 INFORMATION: Freeing texture...
20:57:43.598 INFORMATION: Instantiating texture data...
20:57:43.604 INFORMATION: Creating EGLImageKHR...
20:57:43.605 VERBOSE: EGL image 93 created...
20:57:43.606 VERBOSE: Creating OMX_TextureData...
20:57:43.607 INFORMATION: Instantiating texture data...
20:57:43.612 INFORMATION: Creating EGLImageKHR...
20:57:43.613 VERBOSE: EGL image 94 created...
20:57:43.614 VERBOSE: Creating OMX_TextureData...
20:57:43.615 INFORMATION: Instantiating texture data...
20:57:43.620 INFORMATION: Creating EGLImageKHR...
20:57:43.621 VERBOSE: EGL image 95 created...
20:57:43.622 VERBOSE: Creating OMX_TextureData...
20:57:43.623 INFORMATION: Instantiating texture data...
20:57:43.628 INFORMATION: Creating EGLImageKHR...
20:57:43.630 VERBOSE: EGL image 96 created...
20:57:43.631 VERBOSE: Creating OMX_TextureData...
20:57:43.632 VERBOSE: Texture generated!
Video codec omx-h264 width 1280 height 720 profile 100 fps 25.000000
20:57:43.660 VERBOSE: Opening audio using OMX...
20:57:43.661 VERBOSE: Using omx:hdmi output device...
Audio codec aac channels 2 samplerate 44100 bitspersample 16
20:57:43.690 VERBOSE: Play
20:57:43.691 VERBOSE: Cleaning textures...
20:57:43.694 VERBOSE: Starting thread.
20:57:43.695 VERBOSE: Decoding thread started.
20:57:43.706 VERBOSE: Setting tex: 0.
V:PortSettingsChanged: 1280x720@25.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
qml: Position: 0.
20:57:43.740 DEBUG: bool COMXVideo::SetVideoEGL()
20:57:43.755 DEBUG: Trying to enable to output port...
20:57:43.757 VERBOSE: Creating buffers for 4 images.
20:57:43.761 INFORMATION: Buffer created 0xab1084d0
20:57:43.767 INFORMATION: Buffer created 0xab109070
20:57:43.773 INFORMATION: Buffer created 0xab10f300
20:57:43.779 INFORMATION: Buffer created 0xab108eb0
20:57:43.780 VERBOSE: Component renderer: 1193830.
20:57:43.824 VERBOSE: Stop
20:57:43.825 VERBOSE: Waiting for the stop command to finish.
20:57:43.835 VERBOSE: Stopping OMX clock...
20:57:43.842 INFORMATION: Cleaning up...
20:57:43.844 VERBOSE: Closing players...
20:57:43.984 VERBOSE: Freeing texture...
20:57:43.985 INFORMATION: Cleanup done.
20:57:43.986 INFORMATION: Cleaning up...
20:57:43.987 VERBOSE: Closing players...
20:57:43.988 VERBOSE: Freeing texture...
20:57:43.989 INFORMATION: Cleanup done.
20:57:43.990 INFORMATION: Stop command issued.
20:57:43.993 VERBOSE: Seek 1000ms.
```

desert, 2015-04-09:
Yay! I found it. 

If you hit stop, then seek, it crashes.

dimovnike, 2015-04-09:
this simple QML crashes it (about 9 times from 10) the crash happens regardles of autoPlay.

```
import QtQuick 2.0
import QtMultimedia 5.0

Rectangle {
    Video {
        anchors.fill: parent
        muted: true
        autoPlay:true
        source: "resource/video/compressed.mp4"
    }
}
```

the log is as before:

QML debugging is enabled. Only use this in a safe environment.
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
14:14:36.556 VERBOSE: Instantiating player service...
14:14:36.558 VERBOSE: Instantiating QMediaService...
14:14:36.564 DEBUG: Entering: OpenMAXILPlayerControl::OpenMAXILPlayerControl(QObject*).
14:14:36.566 VERBOSE: Metadata update requested.
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
14:14:36.575 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMediaPlayer(QMediaPlayer*).
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
14:14:36.579 DEBUG: Entering: virtual QMediaPlayer::State OpenMAXILPlayerControl::state() const.
14:14:36.580 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
14:14:36.581 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
14:14:36.600 DEBUG: Entering: virtual void OpenMAXILVideoRendererControl::setSurface(QAbstractVideoSurface*).
14:14:36.602 DEBUG: Entering: virtual bool OpenMAXILPlayerControl::isMuted() const.
14:14:36.604 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMuted(bool).

carlonluca, 2015-04-15:
Should be fixed now. Reopen if you still see issues.

closed 2015-04-15 by carlonluca