carlonluca/pot #22: Video colors are degraded when using this library

dimovnike, 2015-04-08:
The colors of the video are degraded. The gradients, which are otherwise shown fine in omxplayer, appear with reduced color depth (gradients are shown as stripes). I tried with QT_QPA_EGLFS_DEBUG=1 and QT says 24bit however I suspect the color depth is even lower. Let me know if pics would help, or if you need other information about this.

dimovnike, 2015-04-08:
can someone confirm that this still fixes the low color depth issues http://thebugfreeblog.blogspot.ro/2012/11/32-bit-color-depth-qt-on-raspberry-pi.html?showComment=1428446883898#c3127471068375151835 ?

dimovnike, 2015-04-08:
also note that QT completely ignores QT_QPA_EGLFS_DEPTH variable

atr0n, 2015-04-08:
You have to set it manually using QSurfaceFormat on your main.cpp. You could something like this:

    QSurfaceFormat curSurface = viewer.format();
    curSurface.setRedBufferSize(8);
    curSurface.setGreenBufferSize(8);
    curSurface.setBlueBufferSize(8);
    curSurface.setAlphaBufferSize(8);
    viewer.setFormat(curSurface);

I think this issue can be closed since it's not related with the fork.
Regards,

carlonluca, 2015-04-08:
Ah yes, that is probably the best way to set color depth. Thank you @atr0n!
Related to the EGL setup however. I'll set the depth of the POC player to 24 bits for reference.

closed 2015-04-08 by carlonluca

dimovnike, 2015-04-09:
yes this solves it! it increases CPU usage with about 5% (on a 720p video). Thank you!