carlonluca/pot #23: Stop a video, then seek, then crash

desert, 2015-04-09:
In the piomxtextures_app if you stop the video, then seek, it crashes. I guess it goes beyond, and applies  to any player doing those commands in that order.

atr0n, 2015-04-09:
It may be due to openmaxilstreamscontrol is not implemented, right?
Maybe playbackState is not sending playbackStateChanged() signal when the video stops and then, Qt tries to seek on closed file.

dimovnike, 2015-04-09:
Ah yes, it might be the seek because where i do mute i do seek() too. Will do more tests and get back.

dimovnike, 2015-04-09:
correction: looks like seek and mute are different bugs, because in one of the tests I only commented mute part in QML and there are no crashes anymore, however after some seek() calls video becomes laggy.

carlonluca, 2015-04-19:
Seems to be fixed. Can someone double check?

closed 2015-04-19 by carlonluca

carlonluca, 2015-04-19:
If you want to also test stability of the seek implementation you can use POCPlayer. I added a --seektest param:

`./piomxtextures_pocplayer --seektest <uri>`

that will perform a never ending seek.

carlonluca, 2015-04-19:
I stopped the test after 3664 seeks. No issue registered. Seek by itself seems to be ok. Reopen if you still get some crash.