carlonluca/pot #24: compiling piomxtextures_qt_driver requires super user permissions

dimovnike, 2015-04-09:
running make in piomxtextures_qt_driver requires super user permissions, looks like it is doing also make install. Below is the error.

mv -f libopenmaxilmediaplayer.so /usr/local/qt-5.4.1/plugins/mediaservice/ 
mv: inter-device move failed: ‘libopenmaxilmediaplayer.so’ to ‘/usr/local/qt-5.4.1/plugins/mediaservice/libopenmaxilmediaplayer.so’; unable to remove target: Permission denied
Makefile:319: recipe for target '/usr/local/qt-5.4.1/plugins/mediaservice/libopenmaxilmediaplayer.so' failed

carlonluca, 2015-04-09:
Sufficient to change the permissions of the Qt dir.

dimovnike, 2015-04-09:
its a bit of security issue  but I'll close it, there are more important bugs now (the one with mute and seek).

closed 2015-04-09 by dimovnike