carlonluca/pot #26: video tremmor (going back and forth)

dimovnike, 2015-05-06:
I sometimes notice a strange thing, video starts going back and forth for a while, it can be seen here https://youtu.be/HUmVEoqYGH0?t=30 (right at this time).

carlonluca, 2015-08-12:
Could you have a look at this one: http://we.tl/a7j9fvdufK.

Nightmare204, 2015-08-19:
@carlonluca The video tremmor can be notice easily when you pause/play continuously one video.

In the tests made to the issue #27 I don't notice this problem anymore when I use the plugin you attached for the RPi1 version and the binaries uploaded as PiOmxTextures 4.5.0.

carlonluca, 2015-08-19:
Are you sure you can't reproduce with 4.5.0 Pi1 version? It should be fixed in Pi2 version (which is affected instead by #27).

Nightmare204, 2015-08-19:
Well.. I'm making some extra test right now, using only the 4.5.0 Pi1 version, and until now I hadn't seen the video tremmor (with plugin compiled from git sources it does)... could it has to do with some specific video format?

Nightmare204, 2015-08-19:
No video tremmor at all. But I guess @dimovnike would be the best to test this. BTW @carlonluca: isn't the link you provided (http://we.tl/a7j9fvdufK) related to the PiOmxTextures 4.5.0 for the RP1 version? That's the one I'm using...

carlonluca, 2015-08-19:
The important thing is that the last binary provided above is OK. That binary is a development build for both versions. No armv7 optimizations.

carlonluca, 2015-08-29:
Version 5.0.0-beta1 should fix this. Reopen if necessary.

closed 2015-08-29 by carlonluca