carlonluca/pot #27: Crash after playing a puased video

lukas-hetzenecker, 2015-08-10:
I'm using POT 4.5.1 for Raspberry Pi 2 and whenever I try to play a paused video the application crashes with the following error:
```/home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXVideo.cpp:81: OMX_ERRORTYPE fill_buffer_done_callback(OMX_HANDLETYPE, OMX_PTR, OMX_BUFFERHEADERTYPE*): Assertion `provider->registerFilledBuffer(pBuffer)' failed.```

The full output is provided here:

```
13:17:21.091 DEBUG: Entering: virtual void OpenMAXILPlayerControl::pause().
13:17:21.093 VERBOSE: Processing post pause()...
13:17:21.093 DEBUG: Entering: void OpenMAXILPlayerControl::pauseInt().
13:17:21.094 VERBOSE: Pause
13:17:21.095 DEBUG: State changed...
13:17:21.113 INFORMATION: Pause command issued.
13:17:21.985 DEBUG: Entering: virtual void OpenMAXILPlayerControl::play().
13:17:21.986 VERBOSE: Deferring play() command...
13:17:22.001 VERBOSE: Processing post play()...
13:17:22.001 DEBUG: Entering: void OpenMAXILPlayerControl::playInt().
13:17:22.001 VERBOSE: Play
13:17:22.001 VERBOSE: Cleaning textures...
13:17:22.002 INFORMATION: Play command issued.
13:17:22.017 DEBUG: State changed...
13:17:22.047 WARNING: Can't register a filled buffer. None is being filled.
qml: started playback!
qmltest: /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXVideo.cpp:81: OMX_ERRORTYPE fill_buffer_done_callback(OMX_HANDLETYPE, OMX_PTR, OMX_BUFFERHEADERTYPE*): Assertion `provider->registerFilledBuffer(pBuffer)' failed.
bash: line 1: 5957 Aborted DISPLAY=':0.0' /opt/qmltest/bin/qmltest
Application finished with exit code 134.
```

The steps for reproducing this bug are as following:

1. Create a VideoOutput and load a video
2. Start the playback (video.play())
3. Pause it (video.pause())
4. Start it again -> this results in a crash

Can this bug be reproduced by someone else?

carlonluca, 2015-08-10:
Did you also try with POCPlayer? Click on the surface to get the controls and try to pause/play the video. Do you get a crash?

lukas-hetzenecker, 2015-08-10:
yes, exactly the same crash happens with the pocplayer

carlonluca, 2015-08-12:
Could you please have a look at this version of the library: http://we.tl/a7j9fvdufK. This is the version for Pi1, so use this with the Pi1 package.

Nightmare204, 2015-08-19:
@carlonluca I made test with several versions: 
- Qt5.4.2 compiling the plugin from sources --- OK
- Qt5.5.1 compiling the plugin from sources --- OK
- Qt5.5.0, using the binaries provided in the blog for RPi 2 -- Crash
- Qt5.5.0, using the binaries provided in the blog for RPi 1, changing the plugin for the one in the link -- OK

So I guess the problem is related to the improvements you made at the latest RPi2 version, and maybe something missing in our compiling environment or target.

lukas-hetzenecker, 2015-08-19:
Thanks for testing @Nightmare204 - i unfortunately hadn't time to test the provided binaries yet

carlonluca, 2015-08-19:
Yes, only the optimised version should be bugged.

carlonluca, 2015-08-29:
Version 5.0.0-beta1 should fix this. Reopen if needed.

closed 2015-08-29 by carlonluca