carlonluca/pot #28: Qt freezes when starting video

Nightmare204, 2015-08-19:
I don't know if this is a PiOmxTextures issue or not.
Whenever a video starts (play) QtQuick freezes. This happens by one second at most, but it can notice if you make a loop of a short video.

I guess this has to do with a peek in CPU usage to get the video buffers clean and start again.

If this is something unavoidable at expense of taking a longer time starting the video, which configuration would give a better performance?

Currently I'm using a RPi 2 with 256 Mbts to GPU and overclocked at just 950 MHz

Nightmare204, 2015-08-19:
By the way I'm using htop and also Rpi-monitor and CPU usage seem to be stable, but if there are very short CPU usage peek I'm guessing those can't be seen with these tools

carlonluca, 2015-08-19:
Yes, this is the case because opening the video file is done in the same thread. Fixing this requires some work, and I can't guarantee the result.

carlonluca, 2015-08-29:
Version 5.0.0-beta1 should use a aux gl context to avoid blocking the Qt renderer thread. Let me know if it is the expected behaviour.

Nightmare204, 2015-09-07:
V5.0.0-Beta1_PI1 --- Video Change is really Smooth and doesn't freeze the rendering thread.
V5.0.0-Beta1_PI2 --- Same that with V5.0.0-Beta1_PI1

GREAT WORK. THANKS!

closed 2015-09-07 by Nightmare204