carlonluca/pot #29: Crash when changing video - V5.0.0 RPI2

Nightmare204, 2015-09-07:
I just updated to PiOmxTextures 5.0.0 Beta for RPI 2.

When changing videos I noticed the application started to crash.
This just happens with 2 of the test videos I use.

The secuence is as follows:
- Videos reproduce normal.
- I change to the problem video and it reproduces normal.
- Then I change to another video.
- Again I change to the problem video and it reproduced badly (several frame drop)
- When I change video again the application crashes.

This is the Log related to the problem, the problematic video is the file 720p_CocaCola_Refresh.mov (https://drive.google.com/file/d/0B-P_qa88OEwCSTk3eDRkRUxVM28/view?usp=sharing)

The change of videos source was tested under command and also using the signal onStopped, but with the same result. It doesn't matter the amount of videos in between the reproductions of the problematic video, the first one reproduced normal, the second badly and then the app crashes...

```
10:23:30.619 VERBOSE: [0;37mStopping OMX clock...[0m
10:23:30.620 VERBOSE: [0;37mState changing from STATE_PLAYING to STATE_STOPPED...[0m
10:23:30.621 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::stopInt().[0m
10:23:30.622 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::setFilenameInt(QString).[0m
10:23:30.622 INFORMATION: [0;32mCleaning up...[0m
10:23:30.622 VERBOSE: [0;37mClosing players...[0m
10:23:30.738 VERBOSE: [0;37mMetadata update requested.[0m
10:23:30.739 INFORMATION: [0;32mCleanup done.[0m
10:23:30.740 VERBOSE: [0;37mOpening file:///home/pi/Videos/720p_CocaCola_Refresh.mov...[0m
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/Videos/720p_CocaCola_Refresh.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 512
    compatible_brands: qt  
    encoder         : Lavf54.29.104
  Duration: 00:00:15.00, start: 0.000000, bitrate: 1512 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], 1314 kb/s, 
25 fps, 25 tbr, 25 tbn, 50 tbc (default)
    Metadata:
      handler_name    : DataHandler
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 191 kb/s (default)
    Metadata:
      handler_name    : DataHandler
10:23:30.877 VERBOSE: [0;37mCopy metatada...[0m
10:23:30.878 VERBOSE: [0;37mMetaData[0m
10:23:30.878 VERBOSE: [0;37mKey: major_brand, Value: qt  .[0m
10:23:30.878 VERBOSE: [0;37mKey: minor_version, Value: 512.[0m
10:23:30.878 VERBOSE: [0;37mKey: compatible_brands, Value: qt  .[0m
10:23:30.878 VERBOSE: [0;37mKey: encoder, Value: Lavf54.29.104.[0m
10:23:30.878 VERBOSE: [0;37mInitializing OMX clock...[0m
10:23:30.878 VERBOSE: [0;37mMetadata update requested.[0m
10:23:30.882 VERBOSE: [0;37mOpening video using OMX...[0m
10:23:30.882 VERBOSE: [0;37mUsing deinterlace mode: 0.[0m
10:23:30.882 VERBOSE: [0;37mGenerating texture of size (1280, 720).[0m
10:23:30.882 VERBOSE: [0;37mReusing allocated textures.[0m
Video codec omx-h264 width 1280 height 720 profile 100 fps 25.000000
10:23:30.896 VERBOSE: [0;37mOpening audio using OMX...[0m
10:23:30.897 VERBOSE: [0;37mUsing omx:hdmi output device...[0m
Audio codec aac channels 2 samplerate 44100 bitspersample 16
10:23:30.926 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_STOPPED...[0m
10:23:30.926 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
10:23:30.927 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_PLAYING...[0m
10:23:30.927 VERBOSE: [0;37mStarting thread.[0m
10:23:30.928 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
10:23:30.928 VERBOSE: [0;37mDecoding thread started.[0m
V:PortSettingsChanged: 1280x720@25.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
10:23:30.960 VERBOSE: [0;37mCreating buffers for 4 images.[0m
10:23:30.960 INFORMATION: [0;32mBuffer created 0x6a712d00[0m
10:23:30.961 INFORMATION: [0;32mBuffer created 0x6a712d70[0m
10:23:30.961 INFORMATION: [0;32mBuffer created 0x6a712de0[0m
10:23:30.962 INFORMATION: [0;32mBuffer created 0x6a712e50[0m
10:23:30.962 VERBOSE: [0;37mComponent renderer: 108cbb0.[0m
10:23:31.173 WARNING: [0;33mOne frame couldn't be shown.[0m
10:23:31.214 WARNING: [0;33mOne frame couldn't be shown.[0m
10:23:31.253 WARNING: [0;33mOne frame couldn't be shown.[0m
...
10:23:46.052 WARNING: [0;33mOne frame couldn't be shown.[0m
10:23:46.092 WARNING: [0;33mOne frame couldn't be shown.[0m
10:23:46.132 VERBOSE: [0;37mEOS in FillBufferDone cb.[0m
10:23:46.133 WARNING: [0;33mOne frame couldn't be shown.[0m
10:23:46.142 VERBOSE: [0;37mStopping OMX clock...[0m
10:23:46.143 VERBOSE: [0;37mState changing from STATE_PLAYING to STATE_STOPPED...[0m
10:23:46.146 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::stopInt().[0m
10:23:46.147 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::setFilenameInt(QString).[0m
10:23:46.147 INFORMATION: [0;32mCleaning up...[0m
10:23:46.148 VERBOSE: [0;37mClosing players...[0m
10:23:46.166 WARNING: [0;33mOne frame couldn't be shown.[0m
10:23:46.168 WARNING: [0;33mOne frame couldn't be shown.[0m
10:23:46.268 VERBOSE: [0;37mMetadata update requested.[0m
10:23:46.269 INFORMATION: [0;32mCleanup done.[0m
10:23:46.270 VERBOSE: [0;37mOpening file:///home/pi/Videos/720p_CocaCola.mov...[0m
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/Videos/720p_CocaCola.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 512
    compatible_brands: qt  
    encoder         : Lavf54.29.104
  Duration: 00:00:15.05, start: 
0.000000, bitrate: 1779 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], 1582 kb/s, 25 fps, 25 tbr, 25 tbn, 50 tbc (default)
    Metadata:
      handler_name    : DataHandler
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 192 kb/s (default)
    Metadata:
      handler_name    : DataHandler
10:23:46.370 VERBOSE: [0;37mCopy metatada...[0m
10:23:46.370 VERBOSE: [0;37mMetaData[0m
10:23:46.370 VERBOSE: [0;37mKey: major_brand, Value: qt  .[0m
10:23:46.370 VERBOSE: [0;37mKey: minor_version, Value: 512.[0m
10:23:46.370 VERBOSE: [0;37mKey: compatible_brands, Value: qt  .[0m
10:23:46.370 VERBOSE: [0;37mKey: encoder, Value: Lavf54.29.104.[0m
10:23:46.370 VERBOSE: [0;37mInitializing OMX clock...[0m
10:23:46.371 VERBOSE: [0;37mMetadata update requested.[0m
10:23:46.374 VERBOSE: [0;37mOpening video using OMX...[0m
10:23:46.374 VERBOSE: [0;37mUsing deinterlace mode: 0.[0m
10:23:46.374 VERBOSE: [0;37mGenerating texture of size (1280, 720).[0m
10:23:46.374 VERBOSE: [0;37mReusing allocated textures.[0m
Video codec omx-h264 width 1280 height 720 profile 100 fps 25.000000
10:23:46.388 VERBOSE: [0;37mOpening audio using OMX...[0m
10:23:46.388 VERBOSE: [0;37mUsing omx:hdmi output device...[0m
Audio codec aac channels 2 samplerate 44100 bitspersample 16
10:23:46.416 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_STOPPED...[0m
10:23:46.416 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
10:23:46.416 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_PLAYING...[0m
10:23:46.417 VERBOSE: [0;37mStarting thread.[0m
10:23:46.418 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
10:23:46.418 VERBOSE: [0;37mDecoding thread started.[0m
V:PortSettingsChanged: 1280x720@25.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
10:23:46.445 VERBOSE: [0;37mCreating buffers for 4 images.[0m
10:23:46.446 INFORMATION: [0;32mBuffer created 0x6a50d1f0[0m
10:23:46.446 INFORMATION: [0;32mBuffer created 0x6a55d330[0m
10:23:46.447 INFORMATION: [0;32mBuffer created 0x6a51a350[0m
10:23:46.447 INFORMATION: [0;32mBuffer created 0x6a557d50[0m
10:23:46.447 VERBOSE: [0;37mComponent renderer: 10a7ca0.[0m
10:23:46.448 WARNING: [0;33mNo buffer available in any queue.[0m
QT_Tests: /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXVideo.cpp:401: bool COMXVideo::PortSettingsChanged(): Assertion `data' failed.
```

Nightmare204, 2015-09-07:
Tested with V5.0.0_Beta1_PI1 and also V5.0.0_Beta1_PI2.... It happens in both versions

carlonluca, 2015-09-13:
Sorry, not able to reproduce this. Can you reproduce this with POCPlayer?

Nightmare204, 2015-09-14:
I have just tested it with POCPlayer. It happens just with the POCPlayer included with the binaries of V5.0.0_Beta1_PI2 (But it seems to me like Beta1_PI1 includes an old version of POCPlayer because it lacks of some other problems I reported with V5.0.0)

As I reported, this problem happens with just some videos, but it seems to be a bug because it didn't happen early with same sources.

```bash
 Stop command issued.
08:47:02.185 VERBOSE: Stopped signal received!
08:47:02.185 VERBOSE: Entering: bool OMX_MediaProcessor::setFilenameInt(QString).
08:47:02.185 INFORMATION: Cleaning up...
08:47:02.186 VERBOSE: Closing players...
08:47:02.220 WARNING: One frame couldn't be shown.
08:47:02.221 WARNING: One frame couldn't be shown.
08:47:02.313 VERBOSE: Metadata update requested.
08:47:02.315 INFORMATION: Cleanup done.
08:47:02.315 VERBOSE: Opening file:///home/pi/Videos/720p_CocaCola_Refresh.mov...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/Videos/720p_CocaCola_Refresh.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 512
    compatible_brands: qt  
    encoder         : Lavf54.29.104
  Duration: 00:00:15.00, start: 0.000000, bitrate: 1512 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], 1314 kb/s, 25 fps, 25 tbr, 25 tbn, 50 tbc (default)
    Metadata:
      handler_name    : DataHandler
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 191 kb/s (default)
    Metadata:
      handler_name    : DataHandler
08:47:02.467 VERBOSE: Copy metatada...
08:47:02.467 VERBOSE: MetaData
08:47:02.467 VERBOSE: Key: major_brand, Value: qt  .
08:47:02.467 VERBOSE: Key: minor_version, Value: 512.
08:47:02.468 VERBOSE: Key: compatible_brands, Value: qt  .
08:47:02.468 VERBOSE: Key: encoder, Value: Lavf54.29.104.
08:47:02.468 VERBOSE: Initializing OMX clock...
08:47:02.468 VERBOSE: Metadata update requested.
08:47:02.472 VERBOSE: Opening video using OMX...
08:47:02.472 VERBOSE: Using deinterlace mode: 0.
08:47:02.473 VERBOSE: Generating texture of size (1280, 720).
08:47:02.473 VERBOSE: Reusing allocated textures.
Video codec omx-h264 width 1280 height 720 profile 100 fps 25.000000
08:47:02.489 VERBOSE: Opening audio using OMX...
08:47:02.490 VERBOSE: Using omx:hdmi output device...
Audio codec aac channels 2 samplerate 44100 bitspersample 16
08:47:02.520 VERBOSE: State changing from STATE_STOPPED to STATE_STOPPED...
08:47:04.029 VERBOSE: Setting media source...
08:47:04.030 VERBOSE: Sending play command...
08:47:04.031 VERBOSE: Entering: bool OMX_MediaProcessor::playInt().
08:47:04.031 VERBOSE: State changing from STATE_STOPPED to STATE_PLAYING...
08:47:04.032 VERBOSE: Starting thread.
08:47:04.033 VERBOSE: Decoding thread started.
V:PortSettingsChanged: 1280x720@25.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
08:47:04.065 VERBOSE: Creating buffers for 4 images.
08:47:04.066 INFORMATION: Buffer created 0x242d910
08:47:04.066 INFORMATION: Buffer created 0x242d980
08:47:04.067 INFORMATION: Buffer created 0x242d9f0
08:47:04.067 INFORMATION: Buffer created 0x2401ab0
08:47:04.067 VERBOSE: Component renderer: 242bdf0.
08:47:04.068 WARNING: No buffer available in any queue.
piomxtextures_pocplayer: /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXVideo.cpp:401: bool COMXVideo::PortSettingsChanged(): Assertion `data' failed.
Aborted
```

Nightmare204, 2015-09-14:
@carlonluca Have you tested with the video I shared?

carlonluca, 2015-09-14:
Yes, I tested that video. Not sure why, you can try beta2 and see if something has changed... but I couldn't reproduce so I did nothing specific.

Nightmare204, 2015-09-14:
the crash always gives as final message
```
bool COMXVideo::PortSettingsChanged(): Assertion `data' failed.
```

Could it be that I lack of some library for some codec?...

carlonluca, 2015-09-14:
Video codec is h264, so it is provided by the pi. aac should be in ffmpeg, I provided it.
How much memory are you assigning to the GPU? Can you try beta2? Something changed that may influence that assert.

pavelliavonau, 2015-09-16:
This bug reproduces every time in my case. I've made the test for it. I don't use POCPlayer. So I see that only piomxtextures_pocplayer has changed in beta 2 since beta 1 (5.0.0 Pi2).

Here you can find the sources of my test.
https://yadi.sk/d/7vKfQb-Cj84Po

Here you can find the video files for test (/video/adv1 and /video/adv2)
https://yadi.sk/d/pAbT6D9xiXkgF

In testing first and second playback is normal. During third time there are missing frames with message "WARNING: One frame couldn't be shown.". At the begining of the fourth playback application crashes as it was described above in this comments.

I hope my information will help to fix it.

carlonluca, 2015-09-20:
@pavelliavonau I couldn't reproduce any issue with the files you sent me using your sources. Left playing for more than 10 loops. Can you test using beta2? Please check again the version you downloaded because I created a wrong package and there is now an updated version.

@Nightmare204 Can you provide some sample code to reproduce the issue. pocplayer here does not seem to be crashing but you're not the first one reporting this assertion.

dimovnike, 2015-09-22:
confirmed in Qt-rasp-5.5.0. It can change source only one time, the second time it gives a lot of these in the log (it doesnt stop): 

    05:02:30.693 WARNING: One frame couldn't be shown.

the third time (if i change source while it is displaying the above in the log after 10 seconds) it ends up with:

    05:04:59.684 WARNING: No buffer available in any queue.
qmlscene: /home/luca/project/pi/pi/piomxtextures_src/omxplayer_lib/OMXVideo.cpp:401: bool COMXVideo::PortSettingsChanged(): Assertion `data' failed.
Aborted

dimovnike, 2015-09-22:
on qt5.4.1 / PI2 i can not switch videos even once, it ends up with:

```qml: switching to 1 /boot/public.eldorado/video/hit_gold.mp4
05:10:15.099 DEBUG: Entering: virtual void OpenMAXILPlayerControl::stop().
05:10:15.100 VERBOSE: Stop
05:10:15.100 DEBUG: State changed...
05:10:15.101 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
05:10:15.101 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
05:10:15.102 DEBUG: Entering: virtual void OpenMAXILPlayerControl::stop().
05:10:15.102 VERBOSE: Stop
05:10:15.109 VERBOSE: Stopping OMX clock...
05:10:15.110 INFORMATION: Cleaning up...
05:10:15.110 VERBOSE: Closing players...
05:10:15.145 WARNING: One frame couldn't be shown.
05:10:15.236 VERBOSE: Freeing texture...
05:10:15.236 INFORMATION: Cleanup done.

LOCKS HERE and stays liek this infinitely
```

here I do hot swaping, direct source change.

dimovnike, 2015-09-22:
the code 

```
onStatusChanged: {
if (videoLoop.status == MediaPlayer.Loaded) {
// now new source is fully loaded
loopVideo.play()
}```

is not needed, video starts playing automatically after new source is set, however it still crashes.

dimovnike, 2015-09-22:
tested the same QML on PI1 and the result is the same with both qt5.5.0 and qt5.4.1

dimovnike, 2015-09-22:
please let me know if you need my videos and QML.

carlonluca, 2015-09-22:
I already tried the pocplayer but I couldn't reproduce even with the provided video. Can you provide what is needed to reproduce?

pavelliavonau, 2015-09-22:
Luca, thank you very much for update. Beta 2 works fine in my case now. I'll give a feedback if I find new issues.

dimovnike, 2015-09-22:
piomxtextures_pocplayer in PiOmxTextures_5.0.0-beta1_pi1 / qt5.5 is affected by this bug. The behaviour is almost indetical to what i described only it lets me change 1 more videos before it crashes.

dimovnike, 2015-09-22:
beta2 fixes this. Great job!

closed 2015-09-22 by carlonluca