carlonluca/pot #31: Can't Restart if stopped at beginning

Nightmare204, 2015-09-07:
If I start a video (play) and stop it, immediately after it starts the video stop responding to play command.

```
11:55:51.849 VERBOSE: [0;37mOpening file:///home/pi/Videos/720p_cocacola.mov...[0m
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/Videos/720p_cocacola.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 512
    compatible_brands: 
qt  
    encoder         : Lavf54.29.104
  Duration: 00:00:11.01, start: 0.000000, bitrate: 1503 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], 1304 kb/s, 29.97 fps, 29.97 tbr, 30k tbn, 59.94 tbc (default)
    Metadata:
      handler_name    : DataHandler
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 191 kb/s (default)
    Metadata:
      handler_name    : DataHandler
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
11:55:51.944 VERBOSE: [0;37mCopy metatada...[0m
11:55:51.944 VERBOSE: [0;37mMetaData[0m
11:55:51.944 VERBOSE: [0;37mKey: major_brand, Value: qt  .[0m
11:55:51.944 VERBOSE: [0;37mKey: minor_version, Value: 512.[0m
11:55:51.945 VERBOSE: [0;37mKey: compatible_brands, Value: qt  .[0m
11:55:51.945 VERBOSE: [0;37mKey: encoder, Value: Lavf54.29.104.[0m
11:55:51.945 VERBOSE: [0;37mInitializing OMX clock...[0m
11:55:51.951 VERBOSE: [0;37mOpening video using OMX...[0m
11:55:51.951 VERBOSE: [0;37mUsing deinterlace mode: 0.[0m
11:55:51.951 VERBOSE: [0;37mGenerating texture of size (1280, 720).[0m
11:55:51.952 INFORMATION: [0;32mInstantiating texture data...[0m
11:55:51.977 INFORMATION: [0;32mCreating EGLImageKHR...[0m
11:55:51.990 VERBOSE: [0;37mEGL image 1909 created...[0m
11:55:51.990 VERBOSE: [0;37mCreating OMX_TextureData...[0m
11:55:51.990 INFORMATION: [0;32mInstantiating texture data...[0m
11:55:52.008 INFORMATION: [0;32mCreating EGLImageKHR...[0m
11:55:52.008 VERBOSE: [0;37mEGL image 1910 created...[0m
11:55:52.008 VERBOSE: [0;37mCreating OMX_TextureData...[0m
11:55:52.009 INFORMATION: [0;32mInstantiating texture data...[0m
11:55:52.041 INFORMATION: [0;32mCreating EGLImageKHR...[0m
11:55:52.055 VERBOSE: [0;37mEGL image 1911 created...[0m
11:55:52.055 VERBOSE: [0;37mCreating OMX_TextureData...[0m
11:55:52.056 INFORMATION: [0;32mInstantiating texture data...[0m
11:55:52.074 INFORMATION: [0;32mCreating EGLImageKHR...[0m
11:55:52.074 VERBOSE: [0;37mEGL image 1912 created...[0m
11:55:52.074 VERBOSE: [0;37mCreating OMX_TextureData...[0m
Video codec omx-h264 width 1280 height 720 profile 100 fps 29.970030
11:55:52.095 VERBOSE: [0;37mOpening audio using OMX...[0m
11:55:52.096 VERBOSE: [0;37mUsing omx:both output device...[0m
Audio codec aac channels 2 samplerate 44100 bitspersample 16
11:55:52.126 VERBOSE: [0;37mState changing from STATE_INACTIVE to STATE_STOPPED...[0m
11:55:52.127 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
11:55:52.127 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_PLAYING...[0m
11:55:52.128 VERBOSE: [0;37mStarting thread.[0m
11:55:52.129 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
11:55:52.129 VERBOSE: [0;37mDecoding thread started.[0m
V:PortSettingsChanged: 1280x720@29.97 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
11:55:52.151 VERBOSE: [0;37mCreating buffers for 4 images.[0m
11:55:52.152 INFORMATION: [0;32mBuffer created 0x6d465d20[0m
11:55:52.152 INFORMATION: [0;32mBuffer created 0x6d465e10[0m
11:55:52.153 INFORMATION: [0;32mBuffer created 0x6d465e80[0m
11:55:52.153 INFORMATION: [0;32mBuffer created 0x6d465d90[0m
11:55:52.153 VERBOSE: [0;37mComponent renderer: 1b5b338.[0m
11:55:52.172 VERBOSE: [0;37mMetadata update requested.[0m
11:55:52.172 VERBOSE: [0;37mSetting up for new textures...[0m
qml: 07/09 11:55:53 click Stop
11:55:53.507 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::stopInt().[0m
11:55:53.507 VERBOSE: [0;37mWaiting for the stop command to finish.[0m
11:55:53.514 VERBOSE: [0;37mStopping OMX clock...[0m
11:55:53.517 VERBOSE: [0;37mState changing from STATE_PLAYING to STATE_STOPPED...[0m
11:55:53.517 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_STOPPED...[0m
11:55:53.518 INFORMATION: [0;32mStop command issued.[0m
qml: 07/09 11:55:54 click Play
11:55:54.097 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
11:55:54.097 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_PLAYING...[0m
11:55:54.099 VERBOSE: [0;37mStarting thread.[0m
11:55:54.099 VERBOSE: [0;37mDecoding thread started.[0m
11:55:54.100 INFORMATION: [0;32mSeek to: 00:00:00[0m
11:55:54.117 VERBOSE: [0;37mSeeked to: 00:00:00[0m
qml: 07/09 11:55:54 click Stop
11:55:54.736 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::stopInt().[0m
11:55:54.736 VERBOSE: [0;37mWaiting for the stop command to finish.[0m
11:55:54.743 VERBOSE: [0;37mStopping OMX clock...[0m
11:55:54.745 VERBOSE: [0;37mState changing from STATE_PLAYING to STATE_STOPPED...[0m
11:55:54.745 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_STOPPED...[0m
11:55:54.745 INFORMATION: [0;32mStop command issued.[0m
qml: 07/09 11:55:57 click Play
11:55:57.313 VERBOSE: [0;37mEntering: bool OMX_MediaProcessor::playInt().[0m
11:55:57.313 VERBOSE: [0;37mState changing from STATE_STOPPED to STATE_PLAYING...[0m
11:55:57.314 VERBOSE: [0;37mStarting thread.[0m
11:55:57.314 VERBOSE: [0;37mDecoding thread started.[0m
```
However the logs says "Decoding thread started" but no video is shown.

The only way to get video again is to change the source.

I know start/stop a video in such a stressful way is not meant to be done, but I found the bug testing different things.

Nightmare204, 2015-09-07:
This doesn't happen with PiOmxTextures 4.5.0

carlonluca, 2015-09-07:
You mean that you play() and stop() in two consecutive instructions?

Nightmare204, 2015-09-07:
Kind of.
I'm using buttons, so the instructions are not consecutive in code, but separated for a minimum amount of time. Actually I press play, and 1 sec after I press Stop. These buttons just call play() and stop().
I you want I can upload the QML.

carlonluca, 2015-09-07:
Ok, then a pretty common use case actually. I'll take care of this. Thanks for reporting.

carlonluca, 2015-09-13:
Beta 2 should fix this.

closed 2015-09-13 by carlonluca