carlonluca/pot #32: MediaPlayer.loops property now working - V5.5.0

sinadogru, 2015-09-09:
When I try to loop a video infinitely, after the first play it just stops and do not replay the video.

If you like to mimic this here is the simple code,

import QtQuick 2.0
import QtMultimedia 5.0
import QtQuick.Window 2.0

Window {
    visible: true

    MediaPlayer {
        id: player
        source: "file:///home/pi/Videos/big_buck_bunny_1080p_h264.mov"
        loops: MediaPlayer.Infinite
    }

    VideoOutput {
        source: player
    }

    Component.onCompleted: {
        player.play();
    }
}

And also here is the debug output, 

stdin: is not a tty
QML debugging is enabled. Only use this in a safe environment.
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
12:22:53.879 VERBOSE: Instantiating player service...
12:22:53.881 VERBOSE: Instantiating QMediaService...
12:22:53.884 DEBUG: Entering: OpenMAXILPlayerControl::OpenMAXILPlayerControl(QObject*).
12:22:53.885 VERBOSE: Metadata update requested.
12:22:53.887 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMediaPlayer(QMediaPlayer*).
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
12:22:53.888 DEBUG: Entering: virtual QMediaPlayer::State OpenMAXILPlayerControl::state() const.
12:22:53.888 DEBUG: Entering: virtual QMediaPlayer::MediaStatus OpenMAXILPlayerControl::mediaStatus() const.
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
12:22:53.896 DEBUG: Entering: virtual void OpenMAXILVideoRendererControl::setSurface(QAbstractVideoSurface*).
12:22:53.897 DEBUG: Entering: virtual void OpenMAXILPlayerControl::stop().
12:22:53.897 VERBOSE: Stop
12:22:53.898 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMedia(const QMediaContent&, QIODevice*).
12:22:53.898 DEBUG: Media: file:///home/pi/Videos/big_buck_bunny_1080p_h264.mov.
12:22:53.899 DEBUG: setMedia thread is: 0x529758.
12:22:53.899 VERBOSE: Deferring setMedia()...
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
12:22:53.900 DEBUG: Entering: virtual void OpenMAXILPlayerControl::play().
12:22:53.900 VERBOSE: Deferring play() command...
12:22:53.902 VERBOSE: Processing post setMedia()...
12:22:53.902 DEBUG: Entering: void OpenMAXILPlayerControl::setMediaInt(const QMediaContent&).
12:22:53.902 VERBOSE: Stop
12:22:53.902 VERBOSE: Entering: bool OMX_MediaProcessor::setFilenameInt(QString, OMX_TextureData*&).
12:22:53.902 VERBOSE: Opening file:///home/pi/Videos/big_buck_bunny_1080p_h264.mov...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/Videos/big_buck_bunny_1080p_h264.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 537199360
    compatible_brands: qt  
    creation_time   : 2005-09-28 22:15:55
  Duration: 00:00:10.01, start: 0.000000, bitrate: 1485 kb/s
    Stream #0:0(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 128 kb/s (default)
[... 138 lines cut ...]
12:22:54.377 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.393 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.394 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.409 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.410 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.426 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.427 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.442 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.443 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.459 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.460 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.476 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.477 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.492 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.493 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.509 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.510 WARNING: No decoded frame to show at all. Let' render black.
12:22:54.526 WARNING: No decoded frame to show at all. Let' render black.
12:23:04.479 VERBOSE: Stopping OMX clock...
12:23:04.481 INFORMATION: Cleaning up...
12:23:04.482 VERBOSE: Closing players...
12:23:04.620 VERBOSE: Metadata update requested.
12:23:04.623 VERBOSE: Freeing texture...
12:23:04.624 INFORMATION: Cleanup done.
12:23:04.624 DEBUG: State changed...

carlonluca, 2015-09-10:
I'll take care of this. In the meantime you can loop using the stopped signal.

sinadogru, 2015-09-11:
Alright Thank you,

I have done like it and its working,

onStopped: {
    player.stop();
    player.play();
}

carlonluca, 2015-09-13:
Beta 2 should fix this.

closed 2015-09-13 by carlonluca