carlonluca/pot #33: Replay video makes animation lag

sinadogru, 2015-09-14:
Hey,

on the same screen while playing video and have another elemnt which plays animation get lags when the video replay.

here is a sample code to reproduce;

import QtQuick 2.2
import QtQml 2.0
import QtQuick.Window 2.1
import QtMultimedia 5.0

Rectangle {
    width: 1920
    height: 1080
    id: root
    objectName: "root"
    visible: true
    color: "black"
    property int level: 0

    Video {
        anchors.fill: parent
        id: myVideo

        visible: true

        source: "file:///home/pi/Videos/SampleVideo_1080x720_2mb.mp4"
        autoPlay: true

        fillMode: VideoOutput.Stretch

        onStopped: {
            console.log("Stopped.");
            myVideo.seek(0);
            myVideo.play();
        }

        onStatusChanged: {
            console.log("Status changed to: " + status + ".");
        }
    }
    Rectangle {
        border.width: 2
        border.color: "black"
        color: "lightsteelblue"
        anchors.horizontalCenter: parent.horizontalCenter
        width: 250
        height: text.height + 10
        y:1000
        NotificationBox {
            id:text
            width: 200
            anchors.verticalCenter: parent.verticalCenter
            anchors.horizontalCenter: parent.horizontalCenter
            text: "start ------ abcdefghijklmnopqrtaksdjfkdfjklsdjflksdjfklsjadfkljsad;flasjdlfjasdfjldsdfljf---- end"
        }
    }
}

And here is the NotificationBox;

import QtQuick 2.0

Item {
    id:marqueeText
    height: scrollingText.height
    clip: true
    property int tempX: -width
    property alias text: scrollingText.text
    Text {
        x: tempX
        id:scrollingText
        Behavior on x {
            NumberAnimation { duration: 200 }
        }
    }

//    MouseArea {
//        id:mouseArea
//        anchors.fill: parent
//        onClicked: {
//            tempX = 0;
//            timer.running = true;
//        }
//    }

    Timer {
        id:timer
        interval: 200; running: true; repeat: true
        onTriggered:{
            tempX = tempX + 5
            scrollingText.x = -tempX;

            if( tempX + marqueeText.width > scrollingText.width + parent.width) {
                timer.running = false
                pauseTimer.running = true
            }
        }
    }

    Timer {
        id:pauseTimer
        interval: 500; running: false; repeat: false
        onTriggered: {
//            scrollingText.x = 0
            tempX = -parent.width
            timer.running = true
        }
    }
}

You can check this bug with a shorter video easily. 

Thank You,
Sina

carlonluca, 2015-09-14:
What version are you referring to?

keytee, 2015-09-14:
Hello. Same here. Also when changing source. Is there any way to do .source change of a video in async mode? (lattest version, just cloned from the git.)

carlonluca, 2015-09-14:
It is async in the binaries. Since 5.0.0 it was improved.

carlonluca, 2015-09-14:
So you mean the animation interrupts while opening a new media? In that case it is normal as setting the media results in POT reading it, parsing the container, creating players etc...
Some improvements can be found in >= 5.0.0 as this is no more done in the renderer thread. Still I expect not a perfect animation on Pi1, which is single core. Maybe better on Pi2.

carlonluca, 2015-09-14:
Probably a dup of #28.

sinadogru, 2015-09-14:
Hello,

I was out out of for shopping. For example, when I am looping a video, every time when it start over (seek(0)) screen got a bit of lag, for example if I have an animated scrolling text like on billboards, it gets lag.  

Yes I've just viewed issue #28 and it looks like duplicate actually. I was using V4.5.1 and forgot to write it to title. I am sorry.

I am just a bit tired so I will check this V5.0.0-beta on later . Btw I am using POT on Pi2.

carlonluca, 2015-09-14:
In this case yes, testa 5.0.0 beta2.

sinadogru, 2015-09-14:
Alright, I will check it and I will give feedback.

Thank you.

carlonluca, 2015-09-16:
Reopen if needed.

closed 2015-09-16 by carlonluca

sinadogru, 2015-09-18:
Yes beta 2 fixed it. Thank you.