carlonluca/pot #34: Crash when GL error

keytee, 2015-09-14:
Sometimes when QML MediaPlayer object source has changed (load a new video) it drops gl error:

15:07:37.610 VERBOSE: Processing post play()...
15:07:37.611 DEBUG: Entering: void OpenMAXILPlayerControl::playInt().
15:07:37.611 VERBOSE: Play
15:07:37.611 VERBOSE: Cleaning textures...
15:07:37.611 VERBOSE: Entering: bool OMX_MediaProcessor::setFilenameInt(QString, OMX_TextureData*&).
15:07:37.611 INFORMATION: Cleaning up...
15:07:37.611 VERBOSE: Closing players...
15:07:37.612 VERBOSE: Freeing texture...
15:07:37.612 INFORMATION: Cleanup done.
15:07:37.612 VERBOSE: Opening /opt/advinet2/resources/media/3/teszt_videok/1.mp4...
15:07:37.616 VERBOSE: Metadata update requested.
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/opt/advinet2/resources/media/3/teszt_videok/1.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: isommp42
    creation_time   : 2015-08-12 06:56:55
  Duration: 00:00:12.08, start: 0.000000, bitrate: 17274 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1920x1080, 17010 kb/s, 29.89 fps, 30 tbr, 90k tbn, 180k tbc (default)
    Metadata:
      creation_time   : 2015-08-12 06:56:55
      handler_name    : VideoHandle
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 256 kb/s (default)
    Metadata:
      creation_time   : 2015-08-12 06:56:55
      handler_name    : SoundHandle
15:07:38.035 VERBOSE: Copy metatada...
15:07:38.036 VERBOSE: MetaData
15:07:38.036 VERBOSE: Key: major_brand, Value: mp42.
15:07:38.036 VERBOSE: Key: minor_version, Value: 0.
15:07:38.036 VERBOSE: Key: compatible_brands, Value: isommp42.
15:07:38.037 VERBOSE: Key: creation_time, Value: 2015-08-12 06:56:55.
15:07:38.037 VERBOSE: Initializing OMX clock...
15:07:38.042 VERBOSE: Opening video using OMX...
15:07:38.043 VERBOSE: Generating texture of size (1920, 1080).
15:07:38.043 INFORMATION: Freeing KHR image...
15:07:38.044 INFORMATION: Freeing texture...
glGetError 0x501
.... crash....

glError is handled in /pi/piomxtextures_src/omx_textureprovider.cpp, function check_gl_error().
const char *error and error = "STRING" crashed...
Changed to this and its fixed:

void check_gl_error() {
   GLenum err = glGetError();

   while (err != GL_NO_ERROR) {

      switch (err) {
      case GL_INVALID_OPERATION:
         log_err("GL error: INVALID_OPERATION");
         break;
      case GL_INVALID_ENUM:
         log_err("GL error: INVALID_ENUM");
         break;
      case GL_INVALID_VALUE:
         log_err("GL error: INVALID_VALUE");
         break;
      case GL_OUT_OF_MEMORY:
         log_err("GL error: OUT_OF_MEMORY");
         break;
      case GL_INVALID_FRAMEBUFFER_OPERATION:
         log_err("GL error: INVALID_FRAMEBUFFER_OPERATION");
         break;
      default:
         return;
      }

      err = glGetError();
   }

}

Thanks for your great work Luca,

Tibor

carlonluca, 2015-09-14:
Oh yes, thanks, err was supposed to be error. Thanks, I'll fix in the next version.

carlonluca, 2015-09-14:
It is actually already fixed in the binaries. You shouldn't get a crash there.

closed 2015-09-14 by carlonluca

keytee, 2015-09-14:
I''m not using the binaries. Recompiled from git.

carlonluca, 2015-09-14:
Yes, git is pretty old actually as you can see from the dates. The binaries have better performance.

keytee, 2015-09-14:
Ahh. Is there any chance that you'll commit the changes to git? I'm compiling directly on an rPi with custom 5.3.2 (debian jessie) QT5

carlonluca, 2015-09-14:
Yes, unfortunately I had problems with my git branches locally.

keytee, 2015-09-14:
Pure (not git) sources will be fine for me :D