carlonluca/pot #35: Crash on cross compiling - V5.0.0 beta 2

sinadogru, 2015-09-18:
Hello,

I am actually do not sure if this is a bug or not but I would try to explain my situation. 

I was using POT 4.5.1 with cross compiling. Everything was great but I had some issues and those was resolved on V5.0.0. So I wanted to try it. I have clean whole POT 4.5.1 from my host and build new SD card with POT 5.0.0 binaries like i did for 4.5.1. 

I setted up host computer same as I did before.

First of all I tested rpi with sample qml codes with qmlscene and with pocplayer. Those was perfectly working. Then I come back to my host computer to cross compile my previous project from 4.5.1 but suddenly it just crashes like 1 second later after the application start to run. 

So I decided to go back to 4.5.1 to understand if my procedure was wrong but it was working perfectly. 

Here is the application output,

stdin: is not a tty
QML debugging is enabled. Only use this in a safe environment.
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
09:31:52.413 VERBOSE: Instantiating player service...
09:31:52.414 VERBOSE: Instantiating QMediaService...
09:31:52.414 VERBOSE: Using deinterlace mode: 0.
09:31:52.415 VERBOSE: Using deinterlace mode: 0.
09:31:52.416 INFORMATION: POT build Aug 27 2015 21:25:04.
09:31:52.419 INFORMATION: Initializing GPU context in media processor...
09:31:52.419 INFORMATION: Initializing buffer provider...
09:31:52.419 VERBOSE: Metadata update requested.
09:31:52.419 ERROR: Failed to get shared OGL context. Please enable it with the proper attr.
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
09:31:52.422 VERBOSE: Entering: bool OMX_MediaProcessor::playInt().
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
09:31:52.430 VERBOSE: Entering: bool OMX_MediaProcessor::stopInt().
09:31:52.430 VERBOSE: Entering: bool OMX_MediaProcessor::setFilenameInt(QString).
09:31:52.430 VERBOSE: Opening file:///home/pi/Videos/SampleVideo_1080x720_2mb.mp4...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/Videos/SampleVideo_1080x720_2mb.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    creation_time   : 1970-01-01 00:00:00
    encoder         : Lavf53.24.2
  Duration: 00:00:13.50, start: 0.000000, bitrate: 1248 kb/s
    Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p, 1280x720 [SAR 1:1 DAR 16:9], 862 kb/s, 25 fps, 25 tbr, 12800 tbn, 50 tbc (default)
    Metadata:
      creation_time   : 1970-01-01 00:00:00
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, 5.1, fltp, 381 kb/s (default)
    Metadata:
      creation_time   : 1970-01-01 00:00:00
      handler_name    : SoundHandler
QDeclarativeAudio::itemChange
QMediaPlayer::onItemChanged
09:31:52.649 VERBOSE: Copy metatada...
09:31:52.649 VERBOSE: MetaData
09:31:52.649 VERBOSE: Key: major_brand, Value: isom.
09:31:52.650 VERBOSE: Key: minor_version, Value: 512.
09:31:52.650 VERBOSE: Key: compatible_brands, Value: isomiso2avc1mp41.
09:31:52.650 VERBOSE: Key: creation_time, Value: 1970-01-01 00:00:00.
09:31:52.650 VERBOSE: Key: encoder, Value: Lavf53.24.2.
09:31:52.650 VERBOSE: Initializing OMX clock...
09:31:52.651 VERBOSE: Metadata update requested.
09:31:52.656 VERBOSE: Opening video using OMX...
09:31:52.656 VERBOSE: Using deinterlace mode: 0.
09:31:52.657 VERBOSE: Generating texture of size (1280, 720).
09:31:52.657 INFORMATION: Instantiating texture data...
09:31:52.657 INFORMATION: Creating EGLImageKHR...
09:31:52.657 VERBOSE: EGL image 0 created...
09:31:52.657 INFORMATION: Instantiating texture data...
09:31:52.658 INFORMATION: Creating EGLImageKHR...
09:31:52.658 VERBOSE: EGL image 0 created...
eglCreateImageKHR:  failed to create image for buffer (nil) target 12465 error 0x3008
09:31:52.657 ERROR: Failed to create KHR image: 12296.
eglCreateImageKHR:  failed to create image for buffer (nil) target 12465 error 0x3008
09:31:52.658 ERROR: Failed to create KHR image: 12296.
09:31:52.658 INFORMATION: Instantiating texture data...
09:31:52.658 INFORMATION: Creating EGLImageKHR...
09:31:52.658 VERBOSE: EGL image 0 created...
09:31:52.658 INFORMATION: Instantiating texture data...
09:31:52.658 INFORMATION: Creating EGLImageKHR...
09:31:52.658 VERBOSE: EGL image 0 created...
09:31:52.659 VERBOSE: Setting up for new textures...
09:31:52.659 WARNING: Couldn't get a valid texture.
Video codec omx-h264 width 1280 height 720 profile 77 fps 25.000000
09:31:52.681 VERBOSE: Opening audio using OMX...
09:31:52.681 VERBOSE: Using omx:hdmi output device...
eglCreateImageKHR:  failed to create image for buffer (nil) target 12465 error 0x3008
09:31:52.658 ERROR: Failed to create KHR image: 12296.
eglCreateImageKHR:  failed to create image for buffer (nil) target 12465 error 0x3008
09:31:52.658 ERROR: Failed to create KHR image: 12296.
Audio codec aac channels 6 samplerate 48000 bitspersample 16
09:31:52.721 VERBOSE: State changing from STATE_INACTIVE to STATE_STOPPED...
09:31:52.721 VERBOSE: Entering: bool OMX_MediaProcessor::playInt().
09:31:52.721 VERBOSE: State changing from STATE_STOPPED to STATE_PLAYING...
09:31:52.723 VERBOSE: Starting thread.
09:31:52.723 VERBOSE: Entering: bool OMX_MediaProcessor::playInt().
09:31:52.724 VERBOSE: Decoding thread started.
V:PortSettingsChanged: 1280x720@25.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
09:31:52.767 VERBOSE: Creating buffers for 4 images.
bash: line 1:  2813 Segmentation fault      DISPLAY=':0.0' /opt/Piomx5002/bin/Piomx5002
Application finished with exit code 139.

And here is my main.qml:

import QtQuick 2.0
import QtMultimedia 5.0

Item {
    width: 480; height: 320

    MediaPlayer {
        id: player
        source: "file:///home/pi/Videos/SampleVideo_1080x720_2mb.mp4"
        autoPlay: true
        onStopped: { player.seek(0); player.play(); }
    }

    VideoOutput {
        id: video
        anchors.fill: parent
        source: player
        fillMode: VideoOutput.Stretch
    }

    Component.onCompleted: {
        player.play();
    }
}

Well also, if I copy this main.qml and run with qmlscene it is also working with 5.0.0 build.

Also it might help to solve the problem, QtCreator editor underline "import QtMultimedia 5.0" and if I mouseover it, it says;

"Errors while reading typeinfo files: Cannot open /opt/rpi/sysroot/usr/local/Qt-rasp2-5.5.0/qml/QtMultimedia/plugins.qmltypes for reading: No such a file or directory"

But there is plugins.qmltypes file.

Thank you,
Sina

carlonluca, 2015-09-19:
Don't know about Qt Creator, but did you read the note in the blog article about beta1?

carlonluca, 2015-09-19:
I'll temporarily close this. Reopen if setting the shared context is not sufficient.

closed 2015-09-19 by carlonluca

sinadogru, 2015-09-19:
Ah I am sorry to bother you but what is that mean "shared context"?

Also yes, I have just checked your red colored note on your blog, it is also about shared context I guess? but I dont have any idea about how is it possible to enable the global shared OpenGL context?

sinadogru, 2015-09-19:
Also as I see you changed the binaries for beta2. I was trying to use beta2 binaries actually.

dimovnike, 2015-09-19:
for the shared context:  in your QT app, just before calling creating Qapplication (QApplication a(argc, argv) usually in your main.cpp) call 

    QCoreApplication::setAttribute(Qt::AA_ShareOpenGLContexts);

carlonluca, 2015-09-19:
And for the packages I'm sorry but the previous one, as reported by many, was simply wrong, identical to beta1. So I fixed the links.

sinadogru, 2015-09-21:
It is perfectly working thank you so much Luca and Dimovnike