carlonluca/pot #37: .jpg image data V-5.0.0 beta 2

sinadogru, 2015-09-21:
It seems like (if I dont miss some image packages) .jpg formatted images gives error like this,

"QML Image: Invalid image data: qrc:///assets/loby.jpg"

Here is the sample code,

    Image {
        width: 480
        height: 320
        source: "assets/loby.jpg"
    }

sinadogru, 2015-09-21:
btw I have tried with .png and it was working

carlonluca, 2015-09-21:
POT only decodes audio and video, not images. Images should be decoded by Qt. If you are using my build it should be using the system libjpeg. But it is not me doing it.

sinadogru, 2015-09-21:
Thank you, you are right.

clean Rasbian image misses libjpeg.so.62 which is "libjpeg62" package. So I have downloaded and it is now working.

closed 2015-09-21 by sinadogru