carlonluca/pot #38: Audio is not playing V-5.0.0 beta 2

sinadogru, 2015-09-22:
I am not sure but do you guys can have audio playing with qml "Audio" element or with something else? Because I have tried pocplayer and sample codes on the repo with qmlscene and both do not play the audio, I mean there is no any sound. 

I have checked if omxplayer does, and yes, there is sound for it.

Here is the command I tried to play audio with qmlscene,
/usr/local/Qt-rasp2-5.5.0/bin/qmlscene audio_simple.qml file:///home/pi/Videos/SampleVideo_1080x720_2mb.mp4

and for POCPlayer, 
/piomxtextures_pocplayer file:///home/pi/Videos/SampleVideo_1080x720_2mb.mp4

both do not provide sound output over jack.

Am I doing something wrong? or someone else also got this issue?

Sina

sinadogru, 2015-09-22:
Do I need to set something to get volume up maybe?

carlonluca, 2015-09-22:
POT does not play on the jack by default. I plays to the HDMI. Is that working? If you need to play on the jack have a look at 4591c30.

sinadogru, 2015-09-23:
I am not sure if POT plays audio over hdmi, I do not have speakers on my hdmi monitor. But I do need to play over jack, and yes it is working! It is working on POCPlayer, and samples with qmlscene and also for my project!

I am sorry, it was my 2nd duplicate subject, I do not have the habit to check solved subject before submitting one. I will do it before I submit new one.

Thank you very much,
Sina

closed 2015-09-23 by sinadogru