carlonluca/pot #39: video seek() method goes to different video position

dimovnike, 2015-09-23:
please see this log:

```
qml: video pos: 9805
qml: video pos: 9906
qml: video pos: 10006
qml: issuing seek(5000) from QML
17:54:22.761 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setPosition(qint64).
17:54:22.761 VERBOSE: Seek 5000ms.
17:54:22.765 INFORMATION: Seek to: 00:00:05
17:54:22.786 VERBOSE: Seeked to: 00:00:05
qml: video pos: 1321
qml: video pos: 1421
qml: video pos: 1521
qml: video pos: 1628
```
as yo usee, it seeks to an earlier point 1321 instead of 5000. Also when i do seek(0) it does:

```
qml: video pos: 9940
qml: video pos: 10039
17:59:43.520 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setPosition(qint64).
17:59:43.520 VERBOSE: Seek 0ms.
17:59:43.524 INFORMATION: Seek to: 00:00:00
17:59:43.547 VERBOSE: Seeked to: 00:00:00
qml: video pos: -144
qml: video pos: -45
qml: video pos: 54
```

note the minus there, is this normal behaviour? The seek(0) doesn't bother me, but seek to a non zero position breaks some QML logic. 

Also the underseek is different from video to video. On this video it is about 3.5 seconds, we have other videos with 0.5 under seek. (i suspect the bigger the video the biger is underseek but its not proven)

Is this a normal behaviour?

carlonluca, 2015-09-25:
Did you try with omxplayer? I suspect the same happens there.

dimovnike, 2015-09-30:
i tried vlc :) the result is the same, so closing it.

closed 2015-09-30 by dimovnike

carlonluca, 2015-09-30:
Interesting, I thought vlc implemented a better seek... however yes, this is not so weird. I can probably have a look however at the negative values, those makes no sense.