carlonluca/pot #42: Path solve error for explicit local files

Nightmare204, 2015-10-19:
I'm working on a project with my qml files in a qrc, but the video files are local.
That's why I need to pass the source strings as "file://home/pi/Videos/......."

The problem is that in this case the path is trimmed as it can be seen in the next debug output
```
17:09:46.587 DEBUG: Entering: virtual void OpenMAXILPlayerControl::setMedia(const QMediaContent&, QIODevice*).
17:09:46.587 DEBUG: [0x6d13b470] Media: file://home/pi/Videos/720p_ci24.mov.
17:09:46.588 DEBUG: [0x6d13b470] setMedia thread is: 0x554758.
17:09:46.588 VERBOSE: Media status changing from MEDIA_STATUS_NO_MEDIA to MEDIA_STATUS_LOADING...
17:09:46.588 VERBOSE: Entering: bool OMX_MediaProcessor::setFilenameInt(const QString&).
17:09:46.589 DEBUG: [0x6d13b470] virtual void OpenMAXILPlayerControl::play()
17:09:46.589 VERBOSE: Opening file://home/pi/Videos/720p_ci24.mov...
17:09:46.590 DEBUG: [0x6d13b470] virtual void OpenMAXILPlayerControl::play()
17:09:46.609 ERROR: Failed to open source /pi/Videos/720p_ci24.mov.
17:09:46.609 VERBOSE: Media status changing from MEDIA_STATUS_LOADING to MEDIA_STATUS_INVALID_MEDIA...
```

Looking into the sources I found at omx_mediaprocessor.cpp:
```
   // It seems that omxplayer expects path and not local URIs.
   QUrl url(filename);
   if (url.isLocalFile() && filename.startsWith("file://"))
      filename = url.path();
```

To get the correct path I have to double the home directory as follows:
```
"file://home/home/pi/Videos/720p_ci24.mov"
```

It seems to be a minor bug

atr0n, 2015-10-19:
Try with "file:///..." instead of "file://..."

carlonluca, 2015-10-19:
I don't think your URL is valid. You are missing a slash indicating the root, which results in what @atr0n suggested.

closed 2015-10-19 by carlonluca

Nightmare204, 2015-10-19:
I'm sorry, my bad!