carlonluca/pot #44: Video Start position

alanvice, 2015-10-27:
Here is a complete demo to reproduce the bug (expires in 48 hours):
http://expirebox.com/download/2ab22febf1987beda8aa7944e819b01a.html

It seem related to the profile / level used to encode the video file.
As you can see  in the demo ah h264 file encoded with ffmpeg and the parameter "-profile:v high -level 40" causes the video to start skipping the very first part of the video.
Same video encoded with "-profile:v baseline -level 30" seems ok and starts correctly.
Thank you for your work,
hope this helps ;)

carlonluca, 2015-10-27:
Ok, I'm able to reproduce. I'll have a look. Thank you.

carlonluca, 2015-10-30:
You should see the fix in the next release. Please reopen if necessary.

closed 2015-10-30 by carlonluca