carlonluca/pot #45: Video stops and restarts automatically after a f ew seconds

ben80, 2015-11-21:
Hi Luca,

after a few seconds, my videos stop and restart automatically. This repeats indefinitely.
I am using the binaries of POT 5.0.0 (Pi1, wheezy).

Command to run video:
 /usr/local/Qt-rasp-5.5.0/bin/qmlscene /home/pi/piview/lib/samples/video_simple.qml file:///home/pi/piview/lib/samples/big_buck_bunny_1080p_h264.mov

Log, see attachement.
Do you have any idea, why this happens?

Thanks for your help,
Ben
[pot_5.0.0_video_restart_log.txt](https://github.com/carlonluca/pi/files/40913/pot_5.0.0_video_restart_log.txt)
[pot_5.0.0_video_restart_log.txt](https://github.com/carlonluca/pi/files/40914/pot_5.0.0_video_restart_log.txt)

carlonluca, 2015-11-22:
End of media is reached. Can you provide that specific file?

ben80, 2015-11-22:
ok, i did some further tests.
I have no idea why it happened, I did a reboot and now the videos play all the way to the end.
Cannot reproduce it any more.
I think you can close this issue. If i can reproduce it, we can reopen it.

Thanks anyway.

closed 2015-11-22 by carlonluca