carlonluca/pot #46: [question] did you try use other frameworks?

RSATom, 2015-11-30:
like GStreamer/libvlc? If yes, by what reason (reasons) did you prefer omxplayer?

carlonluca, 2015-11-30:
At the time I started none was providing what I needed, including omxplayer. The project uses openmax.

closed 2015-11-30 by carlonluca

RSATom, 2015-11-30:
So it's http://thebugfreeblog.blogspot.it/2013/01/custom-qml-component-for-hardware.html just outdated info? Do you use ffmpeg directly?

carlonluca, 2015-11-30:
No, it is still valid. It says I'm leveraging code from omxplayer, properly patched and integrated. So POT uses openmax and ffmpeg directly, and includes all the related code.

RSATom, 2015-12-01:
Understand. Is this code necessary to have hardware accelerated QML on RPi or it's only related to video playback?