carlonluca/pot #48: MediaPlayer QML type error signal

sinadogru, 2015-12-18:
Hello,

I guess MediaPlayer type is suppose to emit a 'error' signal if it encounters an error. But on POT, for example if the source property contains an invalid media url, while implementation gives a debug output which shows the error like,

08:15:54.262 ERROR: Failed to open source /home/pi.

but on the qml side, error signal is not emitted. Here is a simple code,

    MediaPlayer {
        id: player
        source: "file:///home/pi"
        autoPlay: true
        loops: MediaPlayer.Infinite
        onError: console.log("MediaPlayer.error..");
    }

    VideoOutput {
        id: video
        source: player
        fillMode: VideoOutput.Stretch
        anchors.fill: parent
    }

usual expectation is when the source property is wrong, it would emit an error signal. But as I mentioned above, POT gives a debug output but not emit the signal.

I've also checked on my local machine, which is ubuntu, seems like it is supposed to emit this signal, here is the output for the same qml source file above,

GStreamer; Unable to pause - "file:///home/pi"
GStreamer; Unable to pause - "file:///home/pi"
qml: MediaPlayer.error..
Error: "Resource not found."

sinadogru, 2015-12-18:
UPDATE: and if I am not wrong, player.status also supposed to be MediaPlayer.InvalidMedia, which is not if the given source property was invalid.

LucaCarlon, 2015-12-19:
Yes, not all the signals are implemented. I'll add this as well.

carlonluca, 2015-12-19:
Yes, not all the signals are implemented. I'll add this as well.

alvarolaracano, 2016-02-27:
Hi. Is the property duration and onDurationChanged signal implemented as well??

carlonluca, 2016-02-27:
Probably not. You can file a bug report for that as well if you don't get the signal.