carlonluca/pot #49: Streaming was working before V5.0.0

Nightmare204, 2015-12-29:
Luca, you recently mentioned in the blog that "At the moment the code is pretty close to be ready to play remote streams including https"
http://thebugfreeblog.blogspot.com/2015/11/fully-optimised-raspbian-jessie-with.html?showComment=1450740707138#c3733828862728817809

I remember that in older versions, maybe before V5.0.0, POT was able to handle streamings...
I even checked in some of my old test projects and found some url:

rtsp://192.168.2.27/live.sdp  (A camera I have in my office and I checked with POT)
http://techslides.com/demos/sample-videos/small.3gp (A video file I found looking around)

I lost a lot of info with a Ubuntu reinstall I had to do recently, so I can't try an older version of POT right now, to prove me right

carlonluca, 2015-12-29:
It is very likely you are true, I didn't say it wasn't possible to play remote streams. I'm saying it is not ready. Buffering was implemented but never tested and never included in builds. I never tested anything other than local files.

Nightmare204, 2016-01-12:
Could you guide me where should I look in the code to make the streaming available again?

carlonluca, 2016-01-12:
I'm sorry but not at the moment. I was doing this but I had to stop. I'll probably release something in this sense.

carlonluca, 2016-02-07:
@Nightmare204 Can you please have a look at [this build](http://thebugfreeblog.blogspot.it/2016/02/binaries-for-pot-510-with-remote.html) which includes streaming?

Nightmare204, 2016-02-09:
Hi @carlonluca... streaming for mp4, mov and h264 video formats is working fine with the new build, but there are issues with some video extensions:

We tried with this [small.3gp](http://techslides.com/demos/sample-videos/small.3gp) video, and the streaming started but failed after a coule of frames. The big problem is that even restarting the program with another video (local, streaming and any extension) it failed without showing debug output (it just didn't show anything), and the only solution was to reboot the Raspberry (I guess the GPU gets blocked).
However other .3gp files on streaming worked fine.

I have an IP Camera here that offers two formats of streaming: MJPEG and h264. The last one works perfect on streaming with the new build. On the other hand MJPEG streaming seems to buffer a lot of frames before rendering them and therefore the image is kept still for a few seconds and then it changes fast to try to catch up the actual frame (eventually the streaming seems to get to a halt).
I don't know if this issue is due to my lack of [mpeg-2-license-key/](http://www.raspberrypi.com/mpeg-2-license-key/)

Also as I have been changing a lot the image of the raspberry pi, now I'm unable to reproduce AVI video files (local and streaming), the fps of reproduction are very low and many frames seem to be dropped, but again without debug output. Am I lack of some library?

carlonluca, 2016-02-09:
That 3gp file seems to contain a h263 stream, so you won't be able to decode it with this GPU, I don't think there is a decoder for it. If other worked fine it probably means the others contained h264. 3gp can contain both encodings.

I never tested mjpeg streams. It is probably buffering however. Can you provide one?

Do you have a sample of avi video not working?

davidpez1019, 2016-02-09:
Hi @carlonluca The video https://archive.org/download/test-mpeg/test-mpeg.mpg for streaming reproduce in a similar way to mjpeg stream. At moment I can't provide you a real mjpeg stream.
@Nightmare204  We try using this .avi video [VideoTest.zip](https://github.com/carlonluca/pi/files/123700/VideoTest.zip) for .avi video its reproduce but sometimes suddenly stops.

carlonluca, 2016-02-09:
That does not seem to be mjpeg but mpeg1. It is not the same.

carlonluca, 2016-02-17:
@Nightmare204, @davidpez1019 Are you currently using this build? Even for local files?

Nightmare204, 2016-02-17:
Hi @carlonluca... We tested the build, but we are not using it permanently. This is because we have a lot of work donw in qt5.5.1 and we have to stay in stable comertial version  of Qt, and the last one is that.
I don't know if the git sources have this improvement uploaded, and if so we could compile them in qt5.5.1 and give them a longer test.
However in the tests made to the build we used streaming from cameras, streaming from files and local files, with the results we described before

Nightmare204, 2016-02-17:
@carlonluca  About the mjpeg streamings you can try with any of the ones [here](http://www.mjpeg.net/), for example this [http://80.38.183.149:2000/mjpg/video.mjpg](http://80.38.183.149:2000/mjpg/video.mjpg)

atr0n, 2016-06-30:
I have tried the link that @Nightmare204 post and it's not working. Also I have given a try to an udp multicast signal. Both ended up with the same result: "Failed to open source .".
If I am not wrong, streaming signals was working in old versions, right?
Is there any change to bring it back? Maybe you can point me out to somewhere in order start working on a fix.
Thanks,

carlonluca, 2016-07-03:
Can you try to explain exactly what you are trying to achieve? What version of POT are you using? What stream are you trying to play? How are you trying to play it? Do you have a minimal sample to reproduce?

atr0n, 2016-07-03:
Sure! Two years ago I brought this device: http://www.tbsdtv.com/products/tbs2923-moi-plus.html
It converts the TV signal into a multicast IPTV (powered by tvheadend). It allow me to have IPTV over my local network. Also, I have brought a IPWebcam (which is allocated on my house roof).
I was using an old version of POT and I was allow to watch my tv channels & live camera through a qt program on my TVs (I sent you a youtube link sometime ago which showed my program working, but I don't even know if the link is still alive).
After make a fresh install of Jessie and installing the latest version of POT on github, I get the "**Failed to open source .**" message.

To reproduce the channels, I get all the streaming path inside a ListView and each time I press an up/down key, the media source change to the next/prev link. Then, the listview remain locked until the Video component returns success or failure.
Anyway, even if I load just the Video component inside the main.qml file with a fixed udp link(i.e `udp://@239.0.0.1:1234` or `http://80.38.183.149:2000/mjpg/video.mjpg`) I always get the metioned error.

You could try it your self using VLC streaming by pressing Ctr+S an selecting a local mediasource + udp codification. Note that I use udp instead of http because it takes too long to connect each time I change a channel.
Please, keep in mind that you need to add the multicast route to your pi before trying to reproduce any content as it comes disabled by default: `route add -net 224.0.0.0 netmask 240.0.0.0 eth0`.

Please, let me know if I can do anything to help you (logs, test, etc.). I really miss my homestation.

atr0n, 2016-07-09:
Wow... I should read your blog before writting the above comment! I was testing the non-beta version.
Sorry about that.

closed 2016-07-26 by carlonluca