carlonluca/pot #51: ERROR: Failed to create KHR image: 12300

Djoulez, 2016-01-14:
Hi,

In this simple example, I create/destroy the mediaplayer dynamically. After some times, I always got this error:

09:14:08.076 INFORMATION: Instantiating texture data...
09:14:08.096 INFORMATION: Creating EGLImageKHR...
09:14:08.096 INFORMATION: Instantiating texture data...
09:14:08.532 INFORMATION: Creating EGLImageKHR...
09:14:08.533 INFORMATION: Instantiating texture data...
09:14:08.553 INFORMATION: Creating EGLImageKHR...
09:14:08.553 INFORMATION: Instantiating texture data...
09:14:08.554 INFORMATION: Creating EGLImageKHR...
eglCreateImageKHR:  failed to create image for buffer 0x39 target 12465 error 0x300c
09:14:08.556 ERROR: Failed to create KHR image: 12300.

This example is launched with qmlscene.

Do you have any idea about this ?

[mediaPlayerTest.zip](https://github.com/carlonluca/pi/files/90133/mediaPlayerTest.zip)

sinadogru, 2016-01-14:
It is also happening to me, I was just going to declare an issue about it.

floouuu, 2016-02-03:
I think this has something to do wih running out of video ram. 
To see how much video memory is allocated right now (in bytes) you can run `sudo vcdbg reloc |grep "\[" | awk '{split($0,a," "); print a[12]}' | awk '{split($0,a,","); print a[1]}' | awk '{sum+=$1}END{print sum}'
`
I experienced similar behavior when using the Video component. I wrote a simple video player, that plays urls from a xml file. However, after some time, when gpu memory gets full, I have the same error and the program crashes. 
I Have a Loader whichs loads a QML file where a Video component is included. After each video the memory use of the gpu reduces again, but it is 30 MB short. Which means that with each video that I load there are 30 MB more of gpu memory lost. So I think there has to be a memory leak somewhere.

floouuu, 2016-02-03:
I took a short look, but I guess too short, but I noticed that in `openmaxilplayerservice.cpp` contructor many variables are allocated, but I didn't find where they a freed again. It's just a shot into the blue.

Nightmare204, 2016-02-03:
@floouuu Does the same happens if you reuse the QML Component without reloading it in the Loader?

I have tested long term loop playback using a playlist and I've found this kind of problem is really related to dynamic creation of new video component with a Loader

carlonluca, 2016-02-03:
The fix will be available shortly.

floouuu, 2016-02-03:
@Nightmare204 Thank you for the hint. I am abroad now, but I will test this as soon as I am home again - which will be arround tomorrow evening. I will let you know!
@carlonluca Thank you!

carlonluca, 2016-02-07:
Someone who can test the [new build](http://thebugfreeblog.blogspot.it/2016/02/binaries-for-hardware-accelerated-qt.html)?

Nightmare204, 2016-02-26:
Hi @carlonluca... I'm doing some long test on my code and I'm using POT 5.1.0
Are the fixes for POT 5.2.0 commited already?

LucaCarlon, 2016-02-27:
Sources for 5.2.0 will be committed in a few hours.

carlonluca, 2016-02-27:
Sources for 5.2.0 will be committed in a few hours.

carlonluca, 2016-02-27:
Committed.

Nightmare204, 2016-03-02:
I tested again the 5.6 build and I got some weird behavior:

After several loops (More that 8 hours) between 2 videos of 15 sec the player just stopped with segmentation fault.
If I restart the same player program It fails immediately after starting any video.

```
stdin: is not a tty
QML debugging is enabled. Only use this in a safe environment.
Program Starts
Init - Creating GUI instance
PlayerStandAlone::main::Creating Instance
Init - Executing App
Setting Source to qrc:/qml/VideoPlayer.qml
16:16:19.916 INFO:	 POT build Feb  6 2016 18:15:41.
16:16:19.918 INFO:	 Initializing GPU context in media processor...
16:16:19.919 INFO:	 Initializing buffer provider...
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
Requesting control for org.qt-project.qt.audiorolecontrol/5.6...
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
Calling viewer->showFullScreen()
viewer->status() == QQuickView::Ready
qml: resType =1, localPath =/home/pi/Video/26_5_1080p_bbb_cut.mp4
qml: check isAlreadyPlaying() = false
qml: play function called
qml: load function called
qml: start function called
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/Video/26_5_1080p_bbb_cut.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 512
    compatible_brands: qt  
    encoder         : Lavf56.25.101
  Duration: 00:00:15.04, start: 0.000000, bitrate: 15263 kb/s
    Stream #0:0(eng): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080, 14818 kb/s, 24 fps, 24 tbr, 19200 tbn, 38400 tbc (default)
    Metadata:
      handler_name    : DataHandler
      encoder         : H.264
      timecode        : 00:00:00:00
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, 5.1, fltp, 439 kb/s (default)
    Metadata:
      handler_name    : DataHandler
    Stream #0:2(eng): Data: none (tmcd / 0x64636D74), 0 kb/s
    Metadata:
      handler_name    : DataHandler
      timecode        : 00:00:00:00
16:16:21.366 INFO:	 Instantiating texture data...
16:16:21.371 INFO:	 Instantiating texture data...
16:16:21.377 INFO:	 Instantiating texture data...
16:16:21.382 INFO:	 Instantiating texture data...
bash: line 1:  8943 Segmentation fault      DISPLAY=':0.0' /home/pi/PlayerStandAlone/PlayerStandAlone
```

However if I play the video with omxplayer it reproduces normally. 
I noticed that one of the videos seem to finish in a bad frame, because it generates garbage...

In Other test this video just never ended but rendered black....

IDK if it could be just some error in the video format... @carlonluca If you can check [this video](https://drive.google.com/open?id=0B-P_qa88OEwCWGZ6ZEo5WWtwU3c), maybe you find something on it.

I'd reconverted the video and I'm starting tests again... I will tell you

carlonluca, 2016-03-02:
Can you reproduce with a minimal QML looping the video? Can you run on gdb?

carlonluca, 2016-03-13:
Closing the issue untile someone provides a sample to reproduce. Feel free to reopen.

closed 2016-03-13 by carlonluca

floouuu, 2016-03-24:
Hi, 
I'm sorry for the very late response - work, you know :/
@Nightmare204 I figured out that this only happened when the Video was loaded through a Loader component. When using the Video Component directly, I couldn't discover any issues with the 5.6 image from the post of March 12th 2016. @carlonluca  Video ram also seems to remain stable with your fixes.