carlonluca/pot #52: Duration atribute is not trigerred once changed

alvarolaracano, 2016-02-27:
I have just made a simple program using PiomxTextures. I need to know current media duration in order to trigger an event and this attribute changes asynchronously, so in desktop mode i am using onDurationChanged to dispatch event. But this is no longer working in Piomxtextures.

 
MediaPlayer {
        id: player
        autoPlay: false
        onDurationChanged: {
            console.log('Duration changed: ' + player.duration)
                console.log('Finish early')
                timer.interval = (player.duration - 1200)
                timer.start()
    }

atr0n, 2016-02-27:
Maybe you can try to get the duration from metadata and then, make you own countdown counter.

alvarolaracano, 2016-02-27:
In fact i need to trigger an event when the duration metadata is loaded, and i am achieving this using onDurationChanged in desktop mode.

carlonluca, 2016-03-13:
Build 5.2.1 should include the signal. Reopen if needed.

closed 2016-03-13 by carlonluca