carlonluca/pot #55: Cannot compile

MrCustomizer, 2016-04-15:
I've tried compiling the projects for the Pi 1 by using the `prepare_3rdparty.sh` and `prepare_openmaxil_backend.sh` scripts, but I've run into several problems:

`prepare_openmaxil_backend.sh` expects the compiled result of ffmpeg in `../3rdparty/ffmpeg/lib` (line 25). But the files are in `../3rdparty/ffmpeg/ffmpeg_pi1`.

The next problem, I ran into, has to do with the GLES-includes. I get a lot of strange errors of this kind:
http://pastebin.com/iXrcAdMq

Changing the includes from `#include <GLES2/gl2.h>` to `#include <GLES3/gl3.h>` seems to fix this problem, but I have no idea, if that is the correct way to fix this issue.
There is another guy, who had the same problem a few months ago, but he posted his question in the wrong place, so nobody answered:
https://forum.qt.io/topic/62284/qt-5-6-qopengl-library-missing

I'm not able to verify, if my changes are working, as I'm not sure, what the expected build result is and as I don't know yet, how the Qt plugin system works. The script tries to copy some files at the end, but I think the paths at this place are wrong, too:

cp -a build-PiOmxTextures/libPiOmxTextures* PiOmxTextures/lib
cp -a build-PiOmxTextures/piomxtextures/* PiOmxTextures/include
cp -a ../../3rdparty/ffmpeg/include ffmpeg
cp -a ../../3rdparty/ffmpeg/lib ffmpeg

The ffmpeg path probably has to be changed to ../3rdparty/ffmpeg/ffmpeg_pi1. The piomxtextures directories simply doesn't exist after my build finishes.

Can you help me fix these issues?

I haven't understood yet, how to use the Qt plugin. Does Qt automatically use your plugin for video playback if it is present in Qt's plugin directory or do you have to use a special QML-Item? If there is no special QML-item, how does Qt know, which video backend to use?

atr0n, 2016-04-18:
I extract this from a comment "official" blog: http://thebugfreeblog.blogspot.com.es/p/build-procedure-for-piomxtextures.html

`- Next I had to deal with the same gles3 header problem as you had - weird, because the qtbase was configured with -opengl es2. Anyway that's how I've fixed it:
~/raspi/qt5pi/include/QtGui/qopengl.h:
//manually undef this guys...
#undef QT_OPENGL_ES_3_1
#undef QT_OPENGL_ES_3
// before the following lines:
# if defined(QT_OPENGL_ES_3_1)
# include
# elif defined(QT_OPENGL_ES_3)
# include
# else
# include
#endif`

Hope it will be usefull.

MrCustomizer, 2016-04-18:
Oh, didn't see that comment. It will be a few days until I will have time to test it, but I will come back here, when I've tried it and report, if this is working.

Thanks in advance for that hint.

MrCustomizer, 2016-04-22:
@atr0n I seem to be able to compile the project with your advice and the changes in the referenced pull request. Thanks a lot.

But now I'm running into the next problem. The script prepare_openmaxil_backend.sh tries to copy some files after the compile finishes, but does not find the paths:

cp: cannot stat 'build-PiOmxTextures/libPiOmxTextures*': No such file or directory
cp: cannot stat 'build-PiOmxTextures/piomxtextures/*': No such file or directory

I'm out of ideas, again. Is somebody able to help?

carlonluca, 2016-04-23:
prepare_openmaxil_backend.sh is old and not used anymore. Unless you are trying to build an old version follow the latest suggested procedure: http://thebugfreeblog.blogspot.it/p/build-procedure-for-piomxtextures.html. It is far simpler now.

MrCustomizer, 2016-04-23:
Oh, ok. I read the instructions, but I didn't realize the script is outdated. So i should build the plugin directly with qmake, correct? Which subproject is the Qt plugin? Is it piomxtextures_lib or piomxtextures_qt_driver?

atr0n, 2016-04-23:
You should build piomxtextures_qt_driver with qmake (as a normal qt module) and the rebuild the qt multimedia module.

MrCustomizer, 2016-04-23:
Oh, so the Qt Multimedia patch is still needed? Your link mentions at the top, that this is not necessary any more.

atr0n, 2016-04-23:
Opps, I have never done it without patching. I think we need to do a README file including the whole new procedure.

MrCustomizer, 2016-04-23:
I'm having trouble with building the project by using qmake. The system always complains about a missing `.qmake.conf` file. Any advice?

Steps to reproduce:
1. Change to directory `piomxtextures_qt_driver`
2. Run `qmake`
3. Run `make`

Error output:
```
Project ERROR: Project has no top-level .qmake.conf file.
Makefile:42: recipe for target 'sub-mediaplayer-make_first' failed
make: *** [sub-mediaplayer-make_first] Error 3

```

MrCustomizer, 2016-04-23:
OK. It seems, this error can be prevented by running `qmake ../` instead of `qmake`.

carlonluca, 2016-05-06:
So I suppose it is ok to close this.

closed 2016-05-06 by carlonluca

MrCustomizer, 2016-05-08:
Well, in the end I still haven't been able to compile the project. qmake complained about missing qmake.conf files or something like that. But I have moved on for the moment and I am not still trying to compile the project, so I guess it's OK to close the issue.

carlonluca, 2016-05-08:
Add a file named .qmake.conf in piomxtextures_qt_driver/mediaplayer containing the line: "SRC_DIR=<path_to_qt_sources>".

MrCustomizer, 2016-05-08:
Thanks for the advice, I will try that.