carlonluca/pot #58: Also useable for Android devices?

sk2212, 2016-05-01:
Amazing! It seems that it is possible to display 1080p videos on a Raspberry Pi.

So is there a possibility that the code of this project can be used to display HD content also on an Android device?

carlonluca, 2016-05-06:
Qt already has a backend working on many Android devices.

closed 2016-05-06 by carlonluca

sk2212, 2016-05-09:
What? Can you please show me the QT backend which is able to play 1080p media files on an Android device?

carlonluca, 2016-05-09:
The official Qt backend can assuming the device is powerful enough, which means most recent devices.

atr0n, 2016-05-09:
Also, I remember there were some issues back on qt5.4. Video playback only worked via QtQuick and Video object. I don't know if they solved.
Anyway, qt5.7 looks promising for android since they are including material desing guideline and other stuff. Maybe they will improve qtmultimedia.