carlonluca/pot #60: No output on h264 480p Video

metropt, 2016-05-16:
Hi,

I have tried the following video http://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_480p_h264.mov and even if it says it is playing nothing show on the screen: http://pastebin.com/LEFEc6wp

Thank you

metropt, 2016-06-21:
Teste again on last POT build and the problem still exist but now it gives an insuficient resources error.
1080p and 720p works ok.

```
pi@raspberrypi:~ $ ./piomxtextures_pocplayer 
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
18:10:16.755 INFO:       POT build Jun 11 2016 11:17:55.
18:10:16.756 INFO:       bcm_host_init
18:10:16.758 INFO:       Initializing GPU context in media processor...
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
18:10:16.759 INFO:       Initializing buffer provider...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
Requesting control for org.qt-project.qt.audiorolecontrol/5.6...
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
18:10:17.874 VERB:       Can't find extension for .
18:10:17.875 VERB:       Can't find extension for image/vnd.microsoft.icon.
18:10:17.875 VERB:       Can't find extension for image/x-dds.
18:10:17.875 VERB:       Can't find extension for image/x-icns.
18:10:31.487 VERB:       Setting media source...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/big_buck_bunny_1080p_h264.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 537199360
    compatible_brands: qt  
    creation_time   : 2008-05-27 18:40:35
    timecode        : 00:00:00:00
  Duration: 00:09:56.46, start: 0.000000, bitrate: 9725 kb/s
    Stream #0:0(eng): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080, 9282 kb/s, 24 fps, 24 tbr, 2400 tbn, 4800 tbc (default)
    Metadata:
      creation_time   : 2008-05-27 18:40:35
      handler_name    : Apple Alias Data Handler
      encoder         : H.264
    Stream #0:1(eng): Data: none (tmcd / 0x64636D74) (default)
    Metadata:
      creation_time   : 2008-05-27 18:40:35
      handler_name    : Apple Alias Data Handler
      timecode        : 00:00:00:00
    Stream #0:2(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, 5.1, fltp, 437 kb/s (default)
    Metadata:
      creation_time   : 2008-05-27 18:40:35
      handler_name    : Apple Alias Data Handler
18:10:31.675 INFO:       Instantiating texture data...
18:10:31.712 INFO:       Instantiating texture data...
18:10:31.749 INFO:       Instantiating texture data...
18:10:31.785 INFO:       Instantiating texture data...
Video codec omx-h264 width 1920 height 1080 profile 77 fps 24.000000
Audio codec aac channels 6 samplerate 48000 bitspersample 16
18:10:37.105 INFO:       OMXClock using audio as reference
V:PortSettingsChanged: 1920x1080@24.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
18:10:37.192 INFO:       Buffer created 0x67f19f60
18:10:37.193 INFO:       Buffer created 0x67f20380
18:10:37.195 INFO:       Buffer created 0x67f204a0
18:10:37.196 INFO:       Buffer created 0x67f251d0
18:10:52.050 VERB:       Setting media source...
18:10:52.052 INFO:       Stop command issued.
18:10:52.052 VERB:       Stopped signal received!
18:10:52.053 INFO:       Cleaning up...
18:10:52.193 INFO:       Cleanup done.
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/big_buck_bunny_480p_h264.mov':
  Metadata:
    major_brand     : qt  
    minor_version   : 537199360
    compatible_brands: qt  
    creation_time   : 2008-05-27 18:32:32
    timecode        : 00:00:00:00
  Duration: 00:09:56.46, start: 0.000000, bitrate: 3342 kb/s
    Stream #0:0(eng): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 853x480, 2899 kb/s, 24 fps, 24 tbr, 2400 tbn, 4800 tbc (default)
[... 46 lines cut ...]
    Stream #0:1(eng): Data: none (tmcd / 0x64636D74) (default)
    Metadata:
      creation_time   : 2008-05-27 18:36:22
      handler_name    : Apple Alias Data Handler
      timecode        : 00:00:00:00
    Stream #0:2(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, 5.1, fltp, 437 kb/s (default)
    Metadata:
      creation_time   : 2008-05-27 18:36:22
      handler_name    : Apple Alias Data Handler
18:11:13.636 INFO:       Instantiating texture data...
18:11:13.653 INFO:       Instantiating texture data...
18:11:13.671 INFO:       Instantiating texture data...
18:11:13.688 INFO:       Instantiating texture data...
Video codec omx-h264 width 1280 height 720 profile 77 fps 24.000000
Audio codec aac channels 6 samplerate 48000 bitspersample 16
18:11:13.756 virtual OpenMAXILVideoBuffer::~OpenMAXILVideoBuffer()
V:PortSettingsChanged: 1280x720@24.00 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
18:11:16.692 INFO:       Buffer created 0x67f10c80
18:11:16.693 INFO:       Buffer created 0x67ffbe50
18:11:16.693 INFO:       Buffer created 0x67f0b950
18:11:16.694 INFO:       Buffer created 0x67fd58a0
18:11:20.439 INFO:       Stop command issued.
18:11:20.439 VERB:       Stopped signal received!

```

koolkhel, 2018-09-24:
@metropt it didn't like 853 pixels width. It expects even height and width