carlonluca/pot #62: Trying to open rtsp flux from pocplayer

modjo756, 2016-07-10:
Hi i'm trying to open a rtsp video flux from piomxtextures_pocplayer like this : 
./piomxtextures_pocplayer "rtsp://user:password@192.168.xxx.xxx/video.pro1"
this work fine and i can show the video on the rasp (with the binary file provided with your last image qt5.6) ! 
Now, i have downloaded the source of piomxtextures here and compile it to build my own app.
this compile fine, but when i try to open the same media ./piomxtextures_pocplayer "rtsp://user:password@192.168.xxx.xxx/video.pro1"

the app don't open this video ? I suppose you have modified some code on your pc but not on gitHub ? 
could you please help ?

under the log of poc player thet work :+1:
_Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
20:17:29.701 INFO:       POT build Jun 11 2016 11:17:55.
20:17:29.702 INFO:       bcm_host_init
20:17:29.706 INFO:       Initializing GPU context in media processor...
20:17:29.707 INFO:       Initializing buffer provider...                                                                                                                                                                                             
Requesting control for org.qt-project.qt.metadatareadercontrol/5.0...                                                                                                                                                                                
Requesting control for org.qt-project.qt.mediaavailabilitycontrol/5.0...
Requesting control for org.qt-project.qt.mediaplayercontrol/5.0...
Requesting control for org.qt-project.qt.medianetworkaccesscontrol/5.0...
Requesting control for org.qt-project.qt.audiorolecontrol/5.6...
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
20:17:30.788 INFO:       Showing media: rtsp://admin:camCour@192.168.1.20/video.pro1.
20:17:30.792 VERB:       Can't find extension for .
20:17:30.794 VERB:       Can't find extension for image/vnd.microsoft.icon.
20:17:30.796 VERB:       Can't find extension for image/x-dds.
20:17:30.797 VERB:       Can't find extension for image/x-icns.
20:17:30.798 VERB:       Unrecognized media: assuming video.
20:17:30.800 VERB:       Setting media source...
[h264 @ 0x6b11a800] RTP: missed 1 packets
[h264 @ 0x6b11a800] RTP: missed 3 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 2 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 2 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 2 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 3 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] RTP: missed 1 packets                                                                                                                                                                                                            
[h264 @ 0x6b11a800] out of range intra chroma pred mode at 46 41
[h264 @ 0x6b11a800] error while decoding MB 46 41
[h264 @ 0x6b11a800] concealing 3243 DC, 3243 AC, 3243 MV errors in I frame
[h264 @ 0x6b11a800] RTP: missed 669 packets
Input #0, rtsp, from 'rtsp://admin:camCour@192.168.1.20/video.pro1':
  Metadata:
    title           : QStream
    comment         : QStreaming Media
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Baseline), yuvj420p(pc), 1920x1080, 30 fps, 14,08 tbr, 90k tbn, 60 tbc
    Stream #0:1: Audio: pcm_mulaw, 8000 Hz, 1 channels, s16, 64 kb/s
20:17:33.889 INFO:       Instantiating texture data...
20:17:33.929 INFO:       Instantiating texture data...
20:17:33.969 INFO:       Instantiating texture data...
20:17:34.009 INFO:       Instantiating texture data...
Video codec omx-h264 width 1920 height 1080 profile 66 fps 14,083333
Audio codec pcm_mulaw channels 1 samplerate 8000 bitspersample 8
20:17:34.106 INFO:       OMXClock using audio as reference
V:PortSettingsChanged: 1920x1080@14,08 interlace:0 deinterlace:0 anaglyph:0 par:inf layer:0 alpha:255
[NULL @ 0x6b11a800] RTP: missed 309 packets
20:17:34.167 INFO:       Buffer created 0x69c2e290_

the app that didn't work (build from github source) : 
_QML debugging is enabled. Only use this in a safe environment.
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
20:21:53.060 INFO:       POT build Jun 11 2016 11:17:55.
[... 43 lines cut ...]
Requesting control for org.qt-project.qt.audiorolecontrol/5.6...
Requesting control for org.qt-project.qt.videorenderercontrol/5.0...
20:23:11.502 INFORMATION: Showing media: rtsp://admin:camCour@192.168.1.20/video.pro1.
20:23:11.506 VERBOSE: Can't find extension for .
20:23:11.506 VERBOSE: Can't find extension for image/vnd.microsoft.icon.
20:23:11.507 VERBOSE: Can't find extension for image/x-dds.
20:23:11.507 VERBOSE: Can't find extension for image/x-icns.
20:23:11.508 VERBOSE: Can't handle this media, sorry.
20:23:11.508 VERBOSE: test if is video !
20:23:11.508 VERBOSE: Setting media source...
[h264 @ 0x6b01a800] RTP: missed 339 packets
Input #0, rtsp, from 'rtsp://admin:camCour@192.168.1.20/video.pro1':
  Metadata:
    title           : QStream
    comment         : QStreaming Media
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Baseline), yuvj420p(pc), 1920x1080, 30 fps, 30 tbr, 90k tbn, 60 tbc
    Stream #0:1: Audio: pcm_mulaw, 8000 Hz, 1 channels, s16, 64 kb/s
20:23:15.188 INFO:       Instantiating texture data...
20:23:15.228 INFO:       Instantiating texture data...
20:23:15.266 INFO:       Instantiating texture data...
20:23:15.305 INFO:       Instantiating texture data...
Video codec omx-h264 width 1920 height 1080 profile 66 fps 30,000000
Audio codec pcm_mulaw channels 1 samplerate 8000 bitspersample 8
_

carlonluca, 2016-07-11:
If you are referring to 5.3.0-beta1 then yes, it is not yet available on github. Latest tag available is 5.2.2.

modjo756, 2016-07-11:
Ok thanks for your answer. Do you have an idea when you have the time to push this version on gitHub ? Because it works very fine to access ip cam video !!!

carlonluca, 2016-07-11:
I'm about to release a new version. Once it is done I'll upload the sources. A few days probably.

modjo756, 2016-07-11:
Thank's a lot for your answer and your work ! Thanks also for the build of senseHat Qt module, i will test it this week and let you know ! 
Otherwise, for the new release, do you plan to build Qt 5.7 ?

carlonluca, 2016-07-25:
Sources are now on github. Should I close this issue?

modjo756, 2016-07-26:
yes you could ! And thanks for your share !!!

closed 2016-07-26 by carlonluca