carlonluca/pot #64: Compile docs

ivlevdenis, 2016-09-04:
Please write full docs for compile Qt5+piomxtextures.

carlonluca, 2016-09-15:
It is a regular Qt plugin now. Not sure if docs are needed.

tantowiee, 2016-10-04:
how about this error ?
i'm using qt 5.3.2 on raspberry pi, can u help me

Project ERROR: Unknown module(s) in QT: websockets webchannel
Project ERROR: Unknown module(s) in QT: websockets webchannel
Project ERROR: Unknown module(s) in QT: multimedia quick-private
Project ERROR: Unknown module(s) in QT: multimedia quick-private
Project ERROR: Unknown module(s) in QT: multimedia
Project ERROR: Unknown module(s) in QT: multimedia
Project ERROR: Unknown module(s) in QT: qtmultimedia
Project ERROR: Unknown module(s) in QT: multimedia
Error while parsing file /home/pi/workpi/pi-master/piomxtextures_qt_driver/medi$
Project ERROR: Project has no top-level .qmake.conf file.
Project ERROR: Project has no top-level .qmake.conf file.
Error while parsing file /home/pi/workpi/pi-master/piomxtextures_qt_driver/medi$
Project ERROR: Unknown module(s) in QT: multimedia
Project ERROR: Unknown module(s) in QT: multimedia
Error while parsing file /home/pi/workpi/pi-master/piomxtextures_qt_driver/medi$
Project ERROR: Project has no top-level .qmake.conf file.
Error while parsing file /home/pi/workpi/pi-master/piomxtextures_qt_driver/medi$
Project MESSAGE: Building wk1 sample...
Project ERROR: Unknown module(s) in QT: quick-private
Project ERROR: Unknown module(s) in QT: webenginewidgets
Project ERROR: Unknown module(s) in QT: quick-private
Project ERROR: Project has no top-level .qmake.conf file.
Error while parsing file /home/pi/workpi/pi-master/piomxtextures_qt_driver/medi$
Project MESSAGE: Building wk1 sample...
Project ERROR: Unknown module(s) in QT: quick-private
Project ERROR: Unknown module(s) in QT: webenginewidgets
Project ERROR: Unknown module(s) in QT: quick-private
Project ERROR: Project has no top-level .qmake.conf file.
Error while parsing file /home/pi/workpi/pi-master/piomxtextures_qt_driver/medi$
Project MESSAGE: Building wk1 sample...
Project ERROR: Unknown module(s) in QT: quick-private
Project ERROR: Unknown module(s) in QT: webenginewidgets
Project ERROR: Unknown module(s) in QT: quick-private
Project ERROR: Project has no top-level .qmake.conf file.

carlonluca, 2016-10-05:
It seems you are missing many modules and you'll see Qt 5.3 is not supported. Qt >= 5.5 is needed.

closed 2016-10-05 by carlonluca