carlonluca/pot #66: Audio issue when adding webengine in project

Djoulez, 2016-09-27:
Hello,

I got audio issue with POT when I simply add 
QT+= webengine in my pro file

Without webengine added, the sound is correct but as soon as the webengine is added, the sound is awfully noisy.

I tested on both PiOmxTextures_5.4.0-beta1_pi2_jessiel_5.7.0 and POT 5.2.1 

Do you have the same behaviour ?

Here is a small test project that replicates the issue (just replace the path to a video of your choice in main.qml)

[POT_test.tar.gz](https://github.com/carlonluca/pi/files/495427/POT_test.tar.gz)

Kind Regards,

Julien

carlonluca, 2017-01-06:
Please have a look at the new release. It seems audio now works properly in WebEngine. Open again if needed.

Djoulez, 2017-01-09:
Hi,

In fact the issue is not directly in webengine but in MediaPlayer.

Simply by adding QT+= webengine in the pro file (even without using it at all in the project), the sound is broken in the QML MediaPlayer object.

I tested with the new release but the problem remains.

Kind Regards,

Julien

carlonluca, 2017-03-25:
The problem is that two versions of ffmpeg are loaded when QtWebEngine is linked. Next build will fix this.

closed 2017-03-25 by carlonluca