carlonluca/pot #69: Any chances to get piomxtextures_pocplayer_widget work?

sagiadinos, 2018-02-20:
Hello Luca,

I compiled piomxtextures_pocplayer_widget with last binaries (POT 5.7.0 with Qt 5.10.0) and try to start a video. No error occurs, but video did not play and it shows a white screen with a red rect.
The QML based piomxtextures_pocplayer works fine.

Did I miss something or is there no QWidget support?

Kind regards

carlonluca, 2018-02-24:
Yes, no support for widgets. I don't think it makes sense to spend time on implementing this for widgets nowadays, so I don't think I'll do, sorry.

sagiadinos, 2018-03-03:
Hello,
thank you for answering. No need for sorry it is understandable ... unfortunately it makes sense for me. ;).
Kind regards

closed 2018-06-18 by sagiadinos